AaronErhardt/tuxedo-rs #2: How to use?

crisidev, 2022-11-24:
Hello,
I have copied all the files from `tailord/default_configs` to `/etc/tailord`, as well as the DBUS policy and the systemd file.

I have stopped tcc and the standard control center and if I try to run the systemd service, I get this:

```
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com systemd[1]: Starting Tux Tailor hardware control service...
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com systemd[1]: Started Tux Tailor hardware control service.
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }',>
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]: stack backtrace:
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    0: rust_begin_unwind
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:              at ./rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:584:5
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    1: core::panicking::panic_fmt
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:              at ./rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/panicking.rs:142:14
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    2: core::result::unwrap_failed
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:              at ./rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/result.rs:1785:5
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    3: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    4: std::thread::local::LocalKey<T>::with
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    5: <core::pin::Pin<P> as core::future::future::Future>::poll
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    6: tokio::runtime::scheduler::current_thread::Context::enter
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    7: tokio::macros::scoped_tls::ScopedKey<T>::set
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    8: tokio::runtime::scheduler::current_thread::CurrentThread::block_on
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:    9: tokio::runtime::runtime::Runtime::block_on
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:   10: scoped_tls::ScopedKey<T>::set
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:   11: tokio_uring::start
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]:   12: tux_tailord::main
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com tux-tailord[218019]: note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
Nov 24 14:44:03 ua66e08e6882b52.ant.amazon.com systemd[1]: tailord.service: Main process exited, code=exited, status=101/n/a
```

It looks like it is not able to find the keyboard controller and the kernel module is loaded.

crisidev, 2022-11-24:
I am on Ubuntu 22.06, kernel 6.0.0 and on a Tuxedo Pulse.

AaronErhardt, 2022-11-24:
Thanks for the report! I'm currently about to finish my work on the GUI and will soon provide proper instruction on how to use tuxedo-rs.

The intended way of installing tailord is through meson:

```bash
cd tailord
meson _build
cd _build
ninja install
```

Can you try if that makes it work? 
Also, tuxedo-rs only works on Clevo hardware currently. I'm not sure how much Uniwill parts are used by Tuxedo nowadays, but support for that is only half-finished and could also cause the panic.

crisidev, 2022-11-24:
I know Rust decently. Once I am able to start this, I'll see if my pulse if fully supported and add the support if needed. Thanks for the help in the meantime

crisidev, 2022-11-24:
Same result with the meson build. I guess my keyboard type is not supported yet. I'll see if I can add the support myself.

crisidev, 2022-11-24:
So yeah, the problem is that my keyboard sysfs foder is missing all the color / state / brightness configurations: 

```ls /sys/devices/platform/tuxedo_keyboard/
driver  driver_override  input  modalias  power  subsystem  uevent 
```

I can try to simplify the keyboard config for my machine..

AaronErhardt, 2022-11-24:
Interesting, maybe some hardware doesn't support full RGB? 
There isn't great documentation for this either, but this comment might help you a bit: https://github.com/tuxedocomputers/tuxedo-control-center/issues/167#issuecomment-1011378751

crisidev, 2022-11-24:
My HW doesn't have RGB leds. I have fixed it to move to the next problem:

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: DevNotAvailable', tailord/src/fancontrol/mod.rs:31:54
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Seems like it is not recognising the fan at all.

crisidev, 2022-11-24:
Allright, with some hacking I was able to startup the program and read the fan temp and speed.

My device is indeed using Uniwill parts. As soon as I have a bit of time, I'll cut a PR to support my device.

AaronErhardt, 2023-02-17:
Installation instructions have been added, so I think we can close this. I'd be happy to hear if there was any progress on Uniwill support :)

closed 2023-02-17 by AaronErhardt

crisidev, 2023-02-17:
Life and work get in the middle of a lot of things. I have a crude POC that is somewhat working for the Pulse 14 Gen 1. I hope to find the time to polish it and post my findings.