AaronErhardt/tuxedo-rs #7: GUI on Ubuntu 22.04

crisidev, 2023-04-13:
I noticed the gui does not build for Ubuntu 22.04 as it requires a very new version of glibc.

I'll come back with the data around versions ASAP, but in the meantime I was wondering if we can relax the version a little..

AaronErhardt, 2023-04-13:
Well, the app needs libadwaita 1.3 which is what ships with GNOME 44 (so Fedora 38 or Ubuntu 23.04). Downgrading would require quite some work. However, the flatpak build works on any distribution and would be the package format I would like to focus on.

crisidev, 2023-04-13:
Yep, all this come from libadwaita 1.3:

```
Package dependency requirement 'gobject-2.0 >= 2.76' could not be satisfied.
Package 'gobject-2.0' has version '2.72.4', required version is '>= 2.76'
```

```
Package dependency requirement 'gtk4 >= 4.10' could not be satisfied.
Package 'gtk4' has version '4.6.6', required version is '>= 4.10'
```

petrisch, 2023-04-17:
It fails also with the flatpak variant:

``
error: Requested extension org.freedesktop.Sdk.Extension.llvm15 not installed
Error: Der Kindprozess wurde mit Status 1 beendet
``

Which is kind of odd right? Actually I am still on 20.04, which is the first mistake I know :smile: 
Btw. great stuff, really great :rocket:

AaronErhardt, 2023-04-17:
That should be easy to fix. I just forgot to add `flatpak install runtime/org.freedesktop.Sdk.Extension.llvm15/x86_64/22.08` to the installation instructions. Sometimes it's hard to figure out what you need for installing from scratch because you already have things setup on your system.

The README should be updated now.

petrisch, 2023-04-18:
Yeah that did the job, thanks.

Maybe you can also add to exit TCC from the tray icon eg.
Stopping the service itself was not enough for me.

I haven't yet been able to control the backlight and the output messages are not saying much:
```
INFO socket reader: zbus::socket_reader: enter
INFO socket reader: zbus::socket_reader: exit
```
That's basically it. But I haven't looked into it further.
btw. I have a Aura 1, what controls the input you give by the function keys? Those still work when tuxedo-rs runs.

AaronErhardt, 2023-04-18:
I think the output messages are ok, zbus just tends to create a lot of noise. With backlight, do you mean the LEDs on your keyboard? I think this might be caused by an older version of the kernel module. As the release article mentioned, TUXEDO moved their RGB lighting stuff to `/sys/class/led` where it should have been in the first place. I've decided to drop the old interface in favor of supporting the new one, but it required at least version 3.2.0 (more info [here](https://github.com/tuxedocomputers/tuxedo-keyboard)).

It might be that Ubuntu 20.04 doesn't support the latest, but technically I think it should. So maybe updating the driver works for you.