AaronErhardt/tuxedo-rs #18: Make profile directory configurable

xaverdh, 2023-05-30:
Also /var/lib/tailord may be a better default than /etc/tailord, since these files are generated by the tools and not really 'static' configuration.

AaronErhardt, 2023-05-31:
How would you suggest to implement this? Should we make this a compile-time constant, a CLI argument or just try to automatically detect config files from different locations?

xaverdh, 2023-05-31:
> How would you suggest to implement this? Should we make this a compile-time constant, a CLI argument or just try to automatically detect config files from different locations?

I think a cli argument / env variable is more flexible. Detecting config files raises the question of what to do if both exist.

liketechnik, 2024-05-01:
An env variable could (/should?) be named after [the XDG Specification](https://wiki.archlinux.org/title/XDG_Base_Directory#System_directories), that is `XDG_CONFIG_DIRS` (multiple entries, separate by `:`; akin to PATH). 

Although that still does not resolve the question, which location is to be preferred 🤔