AaronErhardt/tuxedo-rs #19: Issue with suspend

xaverdh, 2023-05-31:
When waking from suspend on my Aura Gen1, the service stops working (the fan curve gets reset). I get
``ERROR tailord::fancontrol: Failed reading the current temperature:  `Device not available` `` in the journal.
Atm I work around this by restarting the service after wake up.

AaronErhardt, 2023-05-31:
I can replicate the error messages on my device (which is also a Aura Gen1), yet this seems to be only caused temporarily by the kernel interface. For a few seconds after the suspend, tailord fails to read the temperature from the kernel driver, but after a few seconds it works again. For tailord this does not cause any real problems because when reading the temperature fails, it will just use the latest available information and print the log message you reported.

You can try to run `sudo RUST_LOG="DEBUG" tailord` to see if the runtime actually stops. Does it continue printing temperature data after suspend on your device?

xaverdh, 2023-05-31:
> I can replicate the error messages on my device (which is also a Aura Gen1), yet this seems to be only caused temporarily by the kernel interface. For a few seconds after the suspend, tailord fails [...]

> You can try to run `sudo RUST_LOG="DEBUG" tailord` to see if the runtime actually stops. Does it continue printing temperature data after suspend on your device?

I can see the failures in the logs intertwined with some successful reads (perhaps due to async logging?). After a few seconds the reads indeed succeed, but the fan curve will only be right every other time (ever after waiting for minutes), like in a race condition.

xaverdh, 2023-05-31:
I also see
`` fan_control_loop: tailord::suspend: Wake up message without suspend.``
for what its worth

xaverdh, 2023-05-31:
here is a sample log: [tailord-log.txt](https://github.com/AaronErhardt/tuxedo-rs/files/11614605/tailord-log.txt)

xaverdh, 2023-05-31:
Could also be unrelated of course. Maybe the failing reads have nothing to do with the failure to set the fan

xaverdh, 2023-05-31:
So judging from the log and experimenting with different fan speeds, it looks like indeed `tailord` is somehow receiving wrong / stale data.

xaverdh, 2023-05-31:
I use the nixos module from this [pr](https://github.com/NixOS/nixpkgs/pull/231384), which pulls in the `tuxedo-keyboard` kernel part. It actually builds all the kernel modules from https://github.com/tuxedocomputers/tuxedo-keyboard/.

xaverdh, 2023-05-31:
> I'll look into it. Which version of the driver are you using btw.?

Thanks!
another odd thing I just noticed, is that the delay in the logs is always 1.733s when the data appears to be wrong. Is that delay from `tailord` or from the kernel?

AaronErhardt, 2023-05-31:
Yes, the delay is fine. The algorithm calculates a pressure value that indicates how fast the temperature is changing and how far the current fan speed is behind the value of the fan curve. According to the pressure value, a delay will be calculated that's up to 2s to reduce CPU time when the temperature doesn't change. If the temperature rises quickly, the delay will be very short so tailord can react faster.

xaverdh, 2023-05-31:
it appears that the nixos package is on an outdated version (3.1.4). I will build `3.2.5` and see if that changes anything

xaverdh, 2023-05-31:
> it appears that the nixos package for the kernel module is on an outdated version (3.1.4). I will build `3.2.5` and see if that changes anything

nope, no dice :-(

AaronErhardt, 2023-06-03:
> but the fan curve will only be right every other time (ever after waiting for minutes), like in a race condition.

I've tried to reproduce, but I can't recreate the same behavior. I did the following:

1. Start tailord with debug info in a console
2. Watch the reported temperatures as I run a heavy applications (Blender in my case)
3. Suspend
4. Wake up after a couple of seconds
5. Do the same test as before (again Blender in my case)

Potentially, only waiting for a longer period before the wake-up triggers this behavior. Can you provide instructions to reproduce the behavior on your machine?

xaverdh, 2023-06-04:
I don't have to run any heavy applications. Even with zero load, the fans will spin up after suspend. Also it appears to be independent of the time my system was suspended I think. Maybe the probability changes slightly, not sure.

If I do just steps 1, 3 and 4 that will reproduce with very high chance on my system (though not 100%).

So if you can't reproduce it, then it looks like either a real hardware defect on my end, or maybe we are on different firmware / EC versions?

AaronErhardt, 2023-06-04:
Doesn't sound very good for sure. We could try a few more things to narrow down the problem:

1. Run `watch -n1 "cat /proc/cpuinfo | grep \"^[c]pu MHz\""` to check the clock speed of your cores to make sure the CPU is actually idle after suspend
2. Run `sudo systemctl stop tailord.service` and then `sudo RUST_LOG="DEBUG" tailord` and watch the temperature reporting

This should show us whether

1. the CPU clock speed was the problem after all
2. the temperature reporting is wrong (tailord gets wrong values from the driver)
3. or tailord fails to set the fan speed

xaverdh, 2023-06-07:
> Doesn't sound very good for sure. We could try a few more things to narrow down the problem:
> 
>     1. Run `watch -n1 "cat /proc/cpuinfo | grep \"^[c]pu MHz\""` to check the clock speed of your cores to make sure the CPU is actually idle after suspend
> 

The cpu is indeed idle

>     2. Run `sudo systemctl stop tailord.service` and then `sudo RUST_LOG="DEBUG" tailord` and watch the temperature reporting
> 
> 
> This should show us whether
> 
>     1. the CPU clock speed was the problem after all
> 

nope

>     2. the temperature reporting is wrong (tailord gets wrong values from the driver)
> 

the temperature reported by tailord is correct (it agrees with readings by lm_sensors and is well below the point where the fan should kick in)

>     3. or tailord fails to set the fan speed

I think this is indeed what happens

AaronErhardt, 2023-06-07:
If this is true, this sounds like a driver/hardware issue. Yet, you mentioned at the beginning, that restarting tailord temporarily solves the issue, which would suggest that the connection between tailord and the ioctl interface breaks during suspend, but only partially since it can still read the temperature correctly. Very strange for sure.

Do you have some knowledge of Rust? You could try the `tuxedo-ioctl` crate in this repo to investigate the ioctl interface to the driver and how it reacts to the suspend. Otherwise, I could probably come up with a small test program as well.

xaverdh, 2023-06-08:
> If this is true, this sounds like a driver/hardware issue. Yet, you mentioned at the beginning, that restarting tailord temporarily solves the issue, which would suggest that the connection between [...]

So the issue is that `tailord` receives wrong fan speed readings after suspend. When I run a small test program though (built against the `tuxedo-ioctl` crate as you suggested) it will obtain the correct values even after suspend.

xaverdh, 2023-06-08:
Ok `receives wrong fan speed readings` is probably wrong, since it doesn't actually read the fan speed.. what I meant is that the debug output from [here](https://github.com/AaronErhardt/tuxedo-rs/blob/21db8da9a7a003ef6dec30dd09b4ffe401fa7430/tailord/src/fancontrol/runtime.rs#L32) shows wrong values..

AaronErhardt, 2023-06-08:
So the debug output shows incorrect temperature readings or what is wrong exactly?

xaverdh, 2023-06-08:
The temperature is correct but the reported fan speed is far too low (I think its the value before suspend)

xaverdh, 2023-06-08:
so somehow after suspend the cached value for the fan speed becomes stale I guess

AaronErhardt, 2023-06-08:
That's interesting. Usually, it should only be a matter of time before the fan speed changes and the condition you commented is true. But I assume your fan curve has an area where it is rather horizontal which could cause `tailord` to never update. And probably the driver just resets the fanspeed on its own after the `suspend`.

I think the most reasonable solution would be to reset the stored fan speed value after waking up from suspend. Then, tailord should immediately overwrite the fan speed on wake up.

xaverdh, 2023-06-08:
Something like this?
[refresh-fan-speed.patch](https://github.com/AaronErhardt/tuxedo-rs/files/11689450/refresh-fan-speed.patch)

xaverdh, 2023-06-08:
> Looks good. Do you want to send a PR?

will do!

closed 2023-06-08 by AaronErhardt