AaronErhardt/tuxedo-rs #21: Help request: adding support for InfinityBook Pro 16 Gen8

tonky, 2023-07-29:
Hi, and thanks for the awesome project!
I just got the IBP16G8 and would love to use these libs to set fan curves and CPU profiles.

I have some Rust experience, but on the web API/application side, not the kernel/low level side.

I'm getting errors from system service, and the same is from compiled `tailord`(with a couple of `println!`s added).

The default 'performance' profile was there when i tried TuxedoOS, but i installed Fedora 38 since.

My understanding is that i should look into ` ioctl_read_int!(profs_available, MAGIC_READ_UW, 0x21);` that tries to read profiles, and see in the source code how official 'TCC' does it, and try to replicate that behavior here.

Is this the way? Any tips or advices? Thanks!

```
[user@fedora ~]$ journalctl -xeu tailord.service
Jul 28 09:09:54 fedora.fritz.box tailord[49824]:  INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default">
Jul 28 09:09:54 fedora.fritz.box tailord[49824]:  INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
Jul 28 09:09:55 fedora.fritz.box tailord[49824]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: DevNotAvailable', tailord/src/main.rs:117:92

❯ cargo build && sudo ../target/debug/tailord
   Compiling tuxedo_ioctl v0.2.0 (/home/user/tuxedo-rs/tuxedo_ioctl)
   Compiling tailord v0.2.1 (/home/user/tuxedo-rs/tailord)
    Finished dev [unoptimized + debuginfo] target(s) in 2.03s
 INFO start_runtime: tailord: Starting tailord
 WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
 INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
 INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
UniwillHardware {
    file: File {
        fd: 9,
        path: "/dev/tuxedo_io",
        read: true,
        write: true,
    },
    num_of_fans: 1,
}
Available profiles: 0
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: DevNotAvailable', tailord/src/main.rs:119:92
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

AaronErhardt, 2023-07-30:
Thanks for your report. 

Which version of the driver are you using? In most parts tuxedo-rs tries to mimic the behavior of tccd (the system service for TCC), so technically if it showed profiles there, it should work for tailord as well. I guess your laptop has uniwill hardware and the function that tries to detect the default profile fails somehow: https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/hal/uniwill.rs#L176

I will look more closely at this later, but I might need some help from you because I just have clevo hardware and can't try this code path myself.

tonky, 2023-07-30:
I've attached the driver version below.

I have no idea how to check for the hardware vendor, any ideas? The shiny sticker on the bottom only says about Tuxedo, nothing about clevo or uniwill.

Thank you for helping ,i'll be happy to provide any extra details, feedback, of participate in testing/building/exploring! :)

```
tuxedo-keyboard on  release took 1m25s 
❯ git log -1
commit e1c0d20c6a80dca087d886090f36a0fb6a39e3f0 (HEAD -> release, origin/release)
Merge: 04a19df 2b04cbc
Author: Christoffer Sandberg <cs@tuxedo.de>
Date:   Tue Jul 25 05:49:42 2023 +0000

    Merge v3.2.

8❯ modinfo tuxedo_io
filename:       /lib/modules/6.4.6-200.fc38.x86_64/extra/tuxedo_io.ko.xz
alias:          wmi:ABBC0F6F-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6E-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          acpi*:CLV0001:*
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6B-8EA1-11D1-00A0-C90629100000
license:        GPL
version:        0.3.6
author:         TUXEDO Computers GmbH <tux@tuxedocomputers.com>
description:    Hardware interface for TUXEDO laptops
rhelversion:    9.99
srcversion:     9679ACBC3A0514B163FC49A
depends:        tuxedo_keyboard
retpoline:      Y
name:           tuxedo_io
vermagic:       6.4.6-200.fc38.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        2B:60:14:BE:25:5E:5E:67:66:6B:5E:27:D5:03:F7:05:23:BA:F6:84
sig_hashalgo:   sha512
signature:      95:52:8F:EC:C8:28:AE:68:1D:37:8A:9C:ED:A8:5C:C4:05:79:59:19:
		55:07:44:B8:F5:32:99:47:94:68:F7:D9:C8:2E:C9:4B:47:A1:79:4B:
		64:A9:FB:C7:A6:1E:8F:BE:04:AA:80:A5:84:CD:C4:01:48:6F:7F:24:
		86:BA:B2:59:41:A6:5F:B1:B8:91:7E:12:D8:F6:67:9D:28:4F:94:B3:
		CC:56:A5:BD:03:D4:DA:EE:B4:E6:5A:BF:F7:70:5D:31:54:3D:92:1F:
		BF:9B:3D:9C:02:C9:CB:0C:DA:3D:E4:32:57:C6:E7:F9:36:D3:23:FD:
		6C:C8:82:98:A3:9A:B0:DE:E3:95:C8:BE:9B:44:45:DF:8B:3C:18:4F:
		B0:AE:7E:64:B7:B8:FB:39:66:3F:6A:9B:F4:53:25:F7:41:12:73:0F:
		36:E6:96:E2:B5:7A:2C:DC:D8:68:04:E9:09:C0:71:9B:5A:1F:CD:14:
		76:0B:92:E6:82:B6:B2:2E:1A:3A:28:46:CC:21:23:DE:BE:BE:3D:E7:
		7A:8F:E7:4E:96:6E:7E:C1:7C:97:7C:1D:6F:D8:DE:72:F4:15:4C:AF:
		A0:D1:21:DE:68:CC:7D:6E:F4:D0:52:49:4E:18:18:41:BE:9B:2C:87:
		D6:D5:D2:B1:1D:7D:2B:AC:3F:6E:33:46:5C:7C:F2:40
```

AaronErhardt, 2023-07-30:
Thanks for the info, that should be the latest version which I also have on my Fedora system.

> I have no idea how to check for the hardware vendor, any ideas? The shiny sticker on the bottom only says about Tuxedo, nothing about clevo or uniwill.

Tuxedo doesn't manufacture every components themselves, but buy them either from clevo or uniwill. For example, the keyboard, the motherboard or the fan might be such components. Of course, Tuxedo still puts them together, but it makes working with the driver a bit challenging due to the two vendors under the hood.

To identify the cause, could you run `tailord` with `RUST_BACKTRACE=1` and show me the log?

tonky, 2023-07-30:
```
❯ RUST_BACKTRACE=1 ../target/debug/tailord > /tmp/tailord_backtrace.log
 INFO start_runtime: tailord: Starting tailord
 WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
 INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
 WARN start_runtime: tailord: No tuxedo ioctl interface available: Permission denied (os error 13)
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: FDO(AccessDenied("Request to own name refused by policy"))', tailord/src/main.rs:163:10
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: core::result::unwrap_failed
   3: core::result::Result<T,E>::unwrap
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/result.rs:1076:23
   4: tailord::start_runtime::{{closure}}::{{closure}}
             at ./src/main.rs:149:17
   5: <tracing::instrument::Instrumented<T> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tracing-0.1.37/src/instrument.rs:272:9
   6: tailord::start_runtime::{{closure}}
             at ./src/main.rs:41:1
   7: tokio_uring::runtime::Runtime::block_on::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/runtime/mod.rs:108:17
   8: <tokio_uring::future::PollFn<F> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/future.rs:35:9
   9: <tokio::task::local::RunUntil<T> as core::future::future::Future>::poll::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:923:42
  10: tokio::task::local::LocalSet::with::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:684:13
  11: std::thread::local::LocalKey<T>::try_with
             at /builddir/build/BUILD/rustc-1.71.0-src/library/std/src/thread/local.rs:270:16
  12: std::thread::local::LocalKey<T>::with
             at /builddir/build/BUILD/rustc-1.71.0-src/library/std/src/thread/local.rs:246:9
  13: tokio::task::local::LocalSet::with
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:667:9
  14: <tokio::task::local::RunUntil<T> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:913:9
  15: tokio::task::local::LocalSet::run_until::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:573:19
  16: <core::pin::Pin<P> as core::future::future::Future>::poll
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/future/future.rs:125:9
  17: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:541:57
  18: tokio::runtime::coop::with_budget
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/coop.rs:107:5
  19: tokio::runtime::coop::budget
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/coop.rs:73:5
  20: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:541:25
  21: tokio::runtime::scheduler::current_thread::Context::enter
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:350:19
  22: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:540:36
  23: tokio::runtime::scheduler::current_thread::CoreGuard::enter::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:615:57
  24: tokio::macros::scoped_tls::ScopedKey<T>::set
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/macros/scoped_tls.rs:61:9
  25: tokio::runtime::scheduler::current_thread::CoreGuard::enter
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:615:27
  26: tokio::runtime::scheduler::current_thread::CoreGuard::block_on
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:530:19
  27: tokio::runtime::scheduler::current_thread::CurrentThread::block_on
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:154:24
  28: tokio::runtime::runtime::Runtime::block_on
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/runtime.rs:302:47
[... cut ...]
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/runtime/mod.rs:105:9
  30: tokio_uring::start
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/lib.rs:146:5
  31: tailord::main
             at ./src/main.rs:38:5
  32: core::ops::function::FnOnce::call_once
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

tonky, 2023-07-30:
```
[root@fedora tailord]# RUST_BACKTRACE=1 ../target/debug/tailord 
 INFO start_runtime: tailord: Starting tailord
 WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
 INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
 INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
UniwillHardware {
    file: File {
        fd: 9,
        path: "/dev/tuxedo_io",
        read: true,
        write: true,
    },
    num_of_fans: 1,
}
Available profiles: 0
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: DevNotAvailable', tailord/src/main.rs:119:92
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: core::result::unwrap_failed
   3: core::result::Result<T,E>::unwrap
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/result.rs:1076:23
   4: tailord::start_runtime::{{closure}}::{{closure}}
             at ./src/main.rs:119:47
   5: <tracing::instrument::Instrumented<T> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tracing-0.1.37/src/instrument.rs:272:9
   6: tailord::start_runtime::{{closure}}
             at ./src/main.rs:41:1
   7: tokio_uring::runtime::Runtime::block_on::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/runtime/mod.rs:108:17
   8: <tokio_uring::future::PollFn<F> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/future.rs:35:9
   9: <tokio::task::local::RunUntil<T> as core::future::future::Future>::poll::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:923:42
  10: tokio::task::local::LocalSet::with::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:684:13
  11: std::thread::local::LocalKey<T>::try_with
             at /builddir/build/BUILD/rustc-1.71.0-src/library/std/src/thread/local.rs:270:16
  12: std::thread::local::LocalKey<T>::with
             at /builddir/build/BUILD/rustc-1.71.0-src/library/std/src/thread/local.rs:246:9
  13: tokio::task::local::LocalSet::with
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:667:9
  14: <tokio::task::local::RunUntil<T> as core::future::future::Future>::poll
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:913:9
  15: tokio::task::local::LocalSet::run_until::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/task/local.rs:573:19
  16: <core::pin::Pin<P> as core::future::future::Future>::poll
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/future/future.rs:125:9
  17: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:541:57
  18: tokio::runtime::coop::with_budget
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/coop.rs:107:5
  19: tokio::runtime::coop::budget
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/coop.rs:73:5
  20: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:541:25
  21: tokio::runtime::scheduler::current_thread::Context::enter
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:350:19
  22: tokio::runtime::scheduler::current_thread::CoreGuard::block_on::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:540:36
  23: tokio::runtime::scheduler::current_thread::CoreGuard::enter::{{closure}}
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:615:57
  24: tokio::macros::scoped_tls::ScopedKey<T>::set
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/macros/scoped_tls.rs:61:9
  25: tokio::runtime::scheduler::current_thread::CoreGuard::enter
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:615:27
  26: tokio::runtime::scheduler::current_thread::CoreGuard::block_on
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:530:19
  27: tokio::runtime::scheduler::current_thread::CurrentThread::block_on
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.28.0/src/runtime/scheduler/current_thread.rs:154:24
  28: tokio::runtime::runtime::Runtime::block_on
[... cut ...]
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/runtime/mod.rs:105:9
  30: tokio_uring::start
             at /home/tonky/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-uring-0.4.0/src/lib.rs:146:5
  31: tailord::main
             at ./src/main.rs:38:5
  32: core::ops::function::FnOnce::call_once
             at /builddir/build/BUILD/rustc-1.71.0-src/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

AaronErhardt, 2023-07-30:
Thanks, I'll look at this tomorrow. I will specially compare the behavior of tccd and tailord to see if there's something different.

AaronErhardt, 2023-07-31:
Mhh, so from what I found, tuxedo-rs does the same as TUXEDO's TCC. 

- tccd: https://github.com/tuxedocomputers/tuxedo-control-center/blob/master/src/native-lib/tuxedo_io_lib/tuxedo_io_api.hh#L410
- tailord: https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/hal/uniwill.rs#L140

The only real difference is that tailord has proper error handling instead of returning a boolean to indicate success...

Could you add a new test case [here](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/read.rs#L171) with the following code and show me the log?

```rust
    #[test]
    fn test_uw_profiles() {
        sudo::escalate_if_needed().unwrap();

        let available_profs = uw::profs_available();
        panic!("INFO: {available_profs}");
    }
```

The panic should print something like `Ok(2)` or `Ok(3)` if your hardware supports performance profiles.

tonky, 2023-07-31:
I've added `file`, since it looks like `profs_available` is expecting it. ran with `cargo test`  and provided `sudo` password when asked. here's the output and diff:

```
tuxedo-rs/tuxedo_ioctl on  main [!] is 📦 v0.2.0 via 🦀 v1.71.0 took 2s 
❯ test hal::test::interface ... FAILED

failures:

---- read::test::test_uw_profiles stdout ----
thread 'read::test::test_uw_profiles' panicked at 'INFO: Ok(
    0,
)', tuxedo_ioctl/src/read.rs:181:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

---- hal::test::interface stdout ----
thread 'hal::test::interface' panicked at 'assertion failed: `(left == right)`
  left: `21`,
 right: `100`', tuxedo_ioctl/src/hal/mod.rs:91:9

failures:
    hal::test::interface
    read::test::test_uw_profiles

test result: FAILED. 2 passed; 2 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.91s

test hal::test::interface ... FAILED

failures:

---- read::test::test_uw_profiles stdout ----
thread 'read::test::test_uw_profiles' panicked at 'INFO: Ok(
    0,
)', tuxedo_ioctl/src/read.rs:181:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

---- hal::test::interface stdout ----
thread 'hal::test::interface' panicked at 'assertion failed: `(left == right)`
  left: `21`,
 right: `100`', tuxedo_ioctl/src/hal/mod.rs:91:9

failures:
    hal::test::interface
    read::test::test_uw_profiles
```

```
diff --git a/tuxedo_ioctl/src/read.rs b/tuxedo_ioctl/src/read.rs
index ab55d4c..4f4a1d2 100644
--- a/tuxedo_ioctl/src/read.rs
+++ b/tuxedo_ioctl/src/read.rs
@@ -169,4 +169,15 @@ mod test {
             assert!(cl::hw_check(&file).unwrap() == 1);
         }
     }
+
+    #[test]
+    fn test_uw_profiles() {
+        sudo::escalate_if_needed().unwrap();
+
+        let file = open_device_file().unwrap();
+        assert!(mod_version(&file).unwrap().contains("0.3"));
+
+        let available_profs = uw::profs_available(&file);
+        panic!("INFO: {:#?}", available_profs);
+    }
 }```

AaronErhardt, 2023-07-31:
This should be the relevant part, more specifically the `Ok(0)`:

```
---- read::test::test_uw_profiles stdout ----
thread 'read::test::test_uw_profiles' panicked at 'INFO: Ok(
    0,
)', tuxedo_ioctl/src/read.rs:181:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

It seems like the kernel driver says there are no (zero) available performance profiles. I think you mentioned that they worked in TCC on Tuxedo OS, right? We could also just ignore this problem and not handle performance profiles if they are not available (instead of panicking, which should be a good idea in general), but if it worked with TCC it should also work with tuxedo-rs, of course.

tonky, 2023-07-31:
yeah, well, maybe those are different "profiles", like some predefined runtime ones?

Here's a [screenshot](https://photos.app.goo.gl/2bTmDPJwPRpi4hCY9) of TCC profiles i have, all were present when i ran it, except "tuxprof" that i created manually.

Initially the "TUXEDO Defaults" was enabled, when i ran TCC.

And here's the `json` of them i found on filesystem:
```
❯ cat /etc/tcc/profiles 
[{"name":"TUXEDO Defaults","description":"Edit profile to change behaviour","display":{"brightness":100,"useBrightness":false},"cpu":{"useMaxPerfGov":true,"governor":"powersave","energyPerformancePreference":"balance_performance","noTurbo":false,"onlineCores":20,"scalingMinFrequency":400000,"scalingMaxFrequency":4800000},"webcam":{"status":true,"useStatus":true},"fan":{"useControl":true,"fanProfile":"Balanced","minimumFanspeed":0,"offsetFanspeed":0},"odmProfile":{"name":""},"odmPowerLimits":{"tdpValues":[42,50,90]},"id":"__default_custom_profile__"},{"id":"__default_mobile_custom_profile__","name":"TUXEDO Mobile Default","description":"Edit profile to change behaviour","display":{"brightness":100,"useBrightness":false},"cpu":{"useMaxPerfGov":false,"scalingMaxFrequency":3500000,"governor":"powersave","energyPerformancePreference":"balance_performance","noTurbo":false,"onlineCores":20,"scalingMinFrequency":400000},"webcam":{"status":true,"useStatus":true},"fan":{"useControl":true,"fanProfile":"Balanced","minimumFanspeed":0,"offsetFanspeed":0},"odmProfile":{"name":""},"odmPowerLimits":{"tdpValues":[15,25,50]}},{"name":"tuxprof","description":"Edit profile to change behaviour","display":{"brightness":100,"useBrightness":false},"cpu":{"useMaxPerfGov":true,"governor":"powersave","energyPerformancePreference":"balance_performance","noTurbo":false,"onlineCores":20,"scalingMinFrequency":400000,"scalingMaxFrequency":4800000},"webcam":{"status":true,"useStatus":true},"fan":{"useControl":true,"fanProfile":"Balanced","minimumFanspeed":0,"offsetFanspeed":0},"odmProfile":{"name":""},"odmPowerLimits":{"tdpValues":[22,35,70]},"id":"spuv5ovqa1llkpyk0oc"}]
```

AaronErhardt, 2023-07-31:
I see, performance profiles weren't supported in TCC either `..."odmProfile":{"name":""}...`.

It's a bit confusing because performance profiles are only about the CPU clock (AFAIK) and the regular profiles are for storing the entire configuration for the system service (fan, LEDs and performance profiles).

The way to solve this would simply be to make performance profiles optional and gracefully handle hardware which doesn't support them.

AaronErhardt, 2023-07-31:
@tonky can you try #22? It should ideally fix your problem.

tonky, 2023-07-31:
oh yeah, `tailord` starts now alright! :)

i could use some help with `tailor-gui` though - when i run it with commands from readme, it starts GUI window with "Connection Error" message:

```
flatpak-builder --run flatpak_app tailor_gui/build-aux/com.github.aaronerhardt.Tailor.json tailor_gui
[...]

2023-07-31T20:46:38.069649Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffef4e21e68"}: relm4::component::sync::builder: new
2023-07-31T20:46:38.069674Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffef4e21e68"}: relm4::component::sync::builder: enter
2023-07-31T20:46:38.069771Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffef4e21e68"}: relm4::component::sync::builder: exit
2023-07-31T20:46:38.069786Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffef4e21e68"}: relm4::component::sync::builder: close time.busy=95.9µs time.idle=42.0µs
```

```
tuxedo-rs on  fix-missing-perf-prof [$] via 🦀 v1.71.0 took 9s 
❯ systemctl status tailord                                                                             
● tailord.service - Tux Tailor hardware control service
     Loaded: loaded (/usr/lib/systemd/system/tailord.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Mon 2023-07-31 22:43:46 CEST; 33s ago
   Main PID: 48713 (tailord)
      Tasks: 2 (limit: 38121)
     Memory: 1.9M
        CPU: 387ms
     CGroup: /system.slice/tailord.service
             └─48713 /usr/bin/tailord

Jul 31 22:43:46 fedora.fritz.box systemd[1]: Starting tailord.service - Tux Tailor hardware control service...
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  INFO start_runtime: tailord: Starting tailord
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or director>
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], perfo>
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  INFO start_runtime: tailord: Tailord started
Jul 31 22:43:46 fedora.fritz.box tailord[48713]:  INFO tailord::suspend: Setting up suspend service
Jul 31 22:43:46 fedora.fritz.box systemd[1]: Started tailord.service - Tux Tailor hardware control service.
```

AaronErhardt, 2023-07-31:
So the problem was that the GUI expected performance profiles to work, which should be fixed now. I've also adjusted the reporting of performance profiles to not throw an error if zero profiles are available because an error should rather mean something bad happened (and zero performance profiles is a valid response of the kernel driver).

Can you try again if the UI works now?

tonky, 2023-07-31:
i pulled the latest changes, rebuild `tailord`, removed `flatpak_app` dir and built `tailor_gui` again, and ran.

still the same error, for some reason.
i see your commits, so i'll try debugging it tomorrow:

```
2023-07-31T21:30:19.867739Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffedc467ee8"}: relm4::component::sync::builder: new
2023-07-31T21:30:19.867760Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffedc467ee8"}: relm4::component::sync::builder: enter
2023-07-31T21:30:19.867919Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffedc467ee8"}: relm4::component::sync::builder: exit
2023-07-31T21:30:19.867926Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`") } component="tailor_gui::app::App" id="0x7ffedc467ee8"}: relm4::component::sync::builder: close time.busy=158µs time.idle=29.2µs`

AaronErhardt, 2023-07-31:
I forgot to mention, you probably have to update tailord as well for the GUI to work.

tonky, 2023-08-01:
here's what i did:

1.  pulled commits from `fix-missing-perf-prof` branch
2.  rebuilt `tailord` and launched it
3.  rebuilt `tailor_gui` with `flatpak-builder` and launched it 

Still seeing the error, so assuming it's sometning on my side i'm gonna look into int.

If my steps are wrong please let me know what to change.

```
tuxedo-rs/tailord on  fix-missing-perf-prof [$?] is 📦 v0.2.1 via 🦀 v1.71.0 
$ cargo build && sudo ../target/debug/tailord
[...] (it's running now)

tuxedo-rs on  fix-missing-perf-prof [$?] via 🦀 v1.71.0 took 1m25s
$ flatpak-builder --force-clean --user flatpak_app tailor_gui/build-aux/com.github.aaronerhardt.Tailor.json
[...]
Saving icon /app/share/app-info/icons/flatpak/128x128/com.github.aaronerhardt.Tailor.png
Saving AppStream /app/share/app-info/xmls/com.github.aaronerhardt.Tailor.xml.gz
Done!
Committing stage cleanup to cache
Finishing app

tuxedo-rs on  fix-missing-perf-prof [$?] via 🦀 v1.71.0 took 2m
$ flatpak-builder --run flatpak_app tailor_gui/build-aux/com.github.aaronerhardt.Tailor.json tailor_gui
Fontconfig error: Cannot load config file "/run/host/font-dirs.xml": No such file: /run/host/font-dirs.xml

(tailor_gui:2): Gtk-WARNING **: 09:28:10.304: Unable to acquire the address of the accessibility bus: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: org.freedesktop.DBus.Error.ServiceUnknown. If you are attempting to run GTK without a11y support, GTK_A11Y should be set to 'none'.
2023-08-01T07:28:10.325556Z  INFO perform: zbus::handshake: new
2023-08-01T07:28:10.325582Z  INFO perform: zbus::handshake: enter
2023-08-01T07:28:10.325934Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: new
2023-08-01T07:28:10.325939Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2023-08-01T07:28:10.325951Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2023-08-01T07:28:10.325955Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: close time.busy=12.1µs time.idle=8.74µs
2023-08-01T07:28:10.325965Z  INFO perform:read_command: zbus::handshake: new
2023-08-01T07:28:10.325967Z  INFO perform:read_command: zbus::handshake: enter
2023-08-01T07:28:10.325971Z  INFO perform:read_command: zbus::handshake: exit
2023-08-01T07:28:10.325973Z  INFO perform: zbus::handshake: exit
2023-08-01T07:28:10.326091Z  INFO perform: zbus::handshake: enter
2023-08-01T07:28:10.326094Z  INFO perform:read_command: zbus::handshake: enter
2023-08-01T07:28:10.326107Z  INFO perform:read_command: zbus::handshake: exit
2023-08-01T07:28:10.326110Z  INFO perform:read_command: zbus::handshake: close time.busy=15.9µs time.idle=130µs
2023-08-01T07:28:10.326116Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: new
2023-08-01T07:28:10.326120Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2023-08-01T07:28:10.326127Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2023-08-01T07:28:10.326131Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: close time.busy=6.20µs time.idle=8.60µs
2023-08-01T07:28:10.326137Z  INFO perform:read_command: zbus::handshake: new
2023-08-01T07:28:10.326140Z  INFO perform:read_command: zbus::handshake: enter
2023-08-01T07:28:10.326145Z  INFO perform:read_command: zbus::handshake: exit
2023-08-01T07:28:10.326148Z  INFO perform: zbus::handshake: exit
2023-08-01T07:28:10.326164Z  INFO perform: zbus::handshake: enter
2023-08-01T07:28:10.326167Z  INFO perform:read_command: zbus::handshake: enter
2023-08-01T07:28:10.326173Z  INFO perform:read_command: zbus::handshake: exit
2023-08-01T07:28:10.326176Z  INFO perform:read_command: zbus::handshake: close time.busy=10.1µs time.idle=29.4µs
2023-08-01T07:28:10.326182Z  INFO perform:write_command{command=Begin}: zbus::handshake: new
2023-08-01T07:28:10.326186Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2023-08-01T07:28:10.326191Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2023-08-01T07:28:10.326195Z  INFO perform:write_command{command=Begin}: zbus::handshake: close time.busy=5.34µs time.idle=7.78µs
2023-08-01T07:28:10.326200Z  INFO perform: zbus::handshake: exit
2023-08-01T07:28:10.326204Z  INFO perform: zbus::handshake: close time.busy=485µs time.idle=168µs
2023-08-01T07:28:10.326537Z  INFO socket reader: zbus::socket_reader: new
2023-08-01T07:28:10.326541Z  INFO socket reader: zbus::socket_reader: enter
2023-08-01T07:28:10.326546Z  INFO socket reader: zbus::socket_reader: exit
2023-08-01T07:28:10.326741Z  INFO socket reader: zbus::socket_reader: enter
2023-08-01T07:28:10.326807Z  INFO socket reader: zbus::socket_reader: exit
2023-08-01T07:28:10.327897Z  INFO socket reader: zbus::socket_reader: enter
2023-08-01T07:28:10.327922Z  INFO socket reader: zbus::socket_reader: exit
2023-08-01T07:28:10.328799Z  INFO socket reader: zbus::socket_reader: enter
2023-08-01T07:28:10.328850Z  INFO socket reader: zbus::socket_reader: exit
2023-08-01T07:28:10.329668Z  INFO socket reader: zbus::socket_reader: enter
2023-08-01T07:28:10.329740Z  INFO socket reader: zbus::socket_reader: exit
2023-08-01T07:28:10.329827Z  INFO socket reader: zbus::socket_reader: close time.busy=230µs time.idle=3.06ms
[... cut ...]
2023-08-01T07:28:13.457098Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffc66dfa058"}: relm4::component::sync::builder: enter
2023-08-01T07:28:13.457117Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffc66dfa058"}: relm4::component::sync::builder: exit
2023-08-01T07:28:13.457123Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffc66dfa058"}: relm4::component::sync::builder: close time.busy=19.6µs time.idle=30.1µs
```

tonky, 2023-08-07:
hey Aaron, here's what i found:
1. `tailor-gui` asks `tailord` about current "performance profile" via DBus.
2. 'tailord` looks it up in `PerformanceInterface:::handler` method
3. `PerformanceInterface.self` is `None`, because `main::performance_profile_handle` is `None` because `main` cant' get `default_performance_profile`

so it looks like we should create and provide some runtime `PerformanceProfileRuntimeHandle`, if we can't get one from hardware?

AaronErhardt, 2023-08-09:
@tonky thanks for investigating this further. It seems like I forgot about the part where tailord asks for the default performance profile, which obviously can't work if there's no functionality provided by the kernel interface. 

With the latest commit, the UI should now treat performance profiles properly as optional. If you want, you can try it out. I'm curious to hear whether that fixes the problem for you :)

tonky, 2023-08-10:
Interesting, it finally runs! :)
 - Led control seems to be working(in 'single' mode).
 - Fan control shows [blank window,](https://photos.app.goo.gl/chMdAxGRCsaZi3C16) but i can click inside and a pair of number show up(probably there's some graph for picking temp points).
 - Fan profiles, led profiles and custom profiles can be created, saved and used.
 
So, all in all, besided visual bug with fan curves - seems to be working ,thanks for your help!

Now i'll need to dig into `tccd` code to see how they set [PL1,2,4 for CPU](https://photos.app.goo.gl/bbPUhcUKKuDZofex5) and try adding configuration for managing those.

AaronErhardt, 2023-08-30:
@tonky I've merged the PR linked to this issue now, which automatically closes this issue. Feel free to re-open this PR or create a new one if you encounter any further problems.

PR #22 (2023-08-30): tailord: Gracefully handle missing performance profiles
Fixes #21
closed 2023-08-30 by AaronErhardt