AaronErhardt/tuxedo-rs #27: Keyboard backlight settings fail to set (NixOS)

MGlolenstine, 2023-10-23:
Whenever I turn on my laptop from sleep, the keyboard lights up (I mainly run with it turned off).

If I try to switch to a "None" configuration in the "default" LED profile, I get the following Toast message:
[image]

And the error reported in the console (when running `RUST_LOG="trace" tailor_gui`) is as following:
```
2023-10-23T05:21:52.968777Z  INFO update_with_view{input=UpdateProfiles { profiles: [FullProfileInfo { name: "default", data: ProfileInfo { fans: ["default"], leds: [], performance_profile: None } }], active_profile: "default", fan_profiles: ["default"], led_profiles: ["default", "empty"] } component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: new
2023-10-23T05:21:52.968791Z  INFO update_with_view{input=UpdateProfiles { profiles: [FullProfileInfo { name: "default", data: ProfileInfo { fans: ["default"], leds: [], performance_profile: None } }], active_profile: "default", fan_profiles: ["default"], led_profiles: ["default", "empty"] } component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: enter
2023-10-23T05:21:52.972003Z  INFO update_with_view{input=UpdateProfiles { profiles: [FullProfileInfo { name: "default", data: ProfileInfo { fans: ["default"], leds: [], performance_profile: None } }], active_profile: "default", fan_profiles: ["default"], led_profiles: ["default", "empty"] } component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: exit
2023-10-23T05:21:52.972028Z  INFO update_with_view{input=UpdateProfiles { profiles: [FullProfileInfo { name: "default", data: ProfileInfo { fans: ["default"], leds: [], performance_profile: None } }], active_profile: "default", fan_profiles: ["default"], led_profiles: ["default", "empty"] } component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: close time.busy=3.21ms time.idle=43.4µs
2023-10-23T05:21:52.972071Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.IOError: No such file or directory (os error 2)`") component="tailor_gui::app::App" id="0x7ffeef96c9a8"}: relm4::component::sync::builder: new
2023-10-23T05:21:52.972082Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.IOError: No such file or directory (os error 2)`") component="tailor_gui::app::App" id="0x7ffeef96c9a8"}: relm4::component::sync::builder: enter
2023-10-23T05:21:52.972748Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.IOError: No such file or directory (os error 2)`") component="tailor_gui::app::App" id="0x7ffeef96c9a8"}: relm4::component::sync::builder: exit
2023-10-23T05:21:52.972767Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.IOError: No such file or directory (os error 2)`") component="tailor_gui::app::App" id="0x7ffeef96c9a8"}: relm4::component::sync::builder: close time.busy=664µs time.idle=31.4µs
```

I don't seem to be able to get more than `INFO` logs though.

AaronErhardt, 2023-10-23:
I think the GUI log isn't very helpful to debug this issue. Could you run `sudo journalctl -e -u tailord.service` to check the logs of the system service?

MGlolenstine, 2023-10-23:
This is the logging I get with the `journalctl`
```
Oct 23 07:06:05 notyours tailord[925]:  INFO start_runtime: tailord: Starting tailord
Oct 23 07:06:05 notyours tailord[925]:  WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
Oct 23 07:06:05 notyours tailord[925]:  INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
Oct 23 07:06:05 notyours systemd[1]: Starting Tuxedo Tailor hardware control service...
Oct 23 07:06:05 notyours tailord[925]:  INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
Oct 23 07:06:05 notyours tailord[925]:  INFO start_runtime: tailord: Tailord started
Oct 23 07:06:05 notyours tailord[925]:  INFO tailord::suspend: Setting up suspend service
Oct 23 07:06:06 notyours systemd[1]: Started Tuxedo Tailor hardware control service.
Oct 23 09:08:07 notyours tailord[925]:  WARN dispatch_message{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}:dispatch_method_call{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}:dispatch_method_call_try{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
Oct 23 09:08:07 notyours tailord[925]:  INFO dispatch_message{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}:dispatch_method_call{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}:dispatch_method_call_try{msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.229"))), path: ObjectPath("/com/tux/Tailor"), iface: InterfaceName(Str(Borrowed("com.tux.Tailor.Profiles"))), member: MemberName(Str(Borrowed("Reload"))) }}: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
Oct 23 10:10:20 notyours tailord[925]:  INFO handle_signals{shutdown_sender=broadcast::Sender}: tailord::shutdown: Received a shutdown signal
Oct 23 10:10:20 notyours tailord[925]:  INFO start_runtime: tailord: Shutting down, bye!
Oct 23 10:10:20 notyours systemd[1]: Stopping Tuxedo Tailor hardware control service...
Oct 23 10:10:20 notyours systemd[1]: tailord.service: Main process exited, code=dumped, status=6/ABRT
Oct 23 10:10:20 notyours systemd[1]: tailord.service: Failed with result 'core-dump'.
Oct 23 10:10:20 notyours systemd[1]: Stopped Tuxedo Tailor hardware control service.
Oct 23 10:10:20 notyours systemd[1]: tailord.service: Consumed 3min 30.633s CPU time, no IP traffic.
Oct 23 10:10:20 notyours systemd[1]: Starting Tuxedo Tailor hardware control service...
Oct 23 10:10:20 notyours tailord[17190]:  INFO start_runtime: tailord: Starting tailord
Oct 23 10:10:20 notyours tailord[17190]:  WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
Oct 23 10:10:20 notyours tailord[17190]:  INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
Oct 23 10:10:20 notyours tailord[17190]:  INFO start_runtime: tailord: Connected to Tuxedo ioctl interface with version 0.3.6
Oct 23 10:10:20 notyours tailord[17190]:  INFO start_runtime: tailord: Tailord started
Oct 23 10:10:20 notyours tailord[17190]:  INFO tailord::suspend: Setting up suspend service
Oct 23 10:10:20 notyours systemd[1]: Started Tuxedo Tailor hardware control service.
```
(I restarted the service at 10:10)

AaronErhardt, 2023-10-23:
@MGlolenstine thanks for your help with debugging this! The symlink seems to be created incorrectly indeed, missing the ".json" ending in the target.

I've opened #28 to address the issue. If it's not too much effort for you it'd be great if you could test whether that branch fixes the problem :)

MGlolenstine, 2023-10-23:
I'd gladly test, but I'm unsure on how to test a NixOS module without merging it with nixpkgs manually.

I do have a flake configuration and I'm using Home manager, how would I integrate the PR 8 into my configuration?

And thank you for quick response :)

AaronErhardt, 2023-10-23:
I already thought it might be a bit complicated to test this based on your NixOS module because you'd probably have to install tailord from source.

I was able to reproduce the bug and fix it with the code on the branch, so it should be safe to merge the code. If I understood the NixOS infrastructure on this repo correctly, the NixOs module should receive an update automatically.

MGlolenstine, 2023-10-23:
So if I understand above correctly, I should be fine just running `nix flake update` and rebuilding the system configuration?
I'm on `nixos/nixpkgs/nixos-unstable`.

PR #28 (2023-10-23): Fix default symlink creation
Fixes #27