AaronErhardt/tuxedo-rs #31: "None" in LED backlight doesn't work correctly.

MGlolenstine, 2023-10-24:
My configuration:
```
❯ cat /etc/tailord/active_profile.json
{"fans":["quiet"],"leds":[{"device_name":"platform:tuxedo_keyboard","function":"kbd_backlight","profile":"default"}],"performance_profile":"performance"}⏎                                                           
~ 
❯ cat /etc/tailord/keyboard/default.json 
"None"
```

Whenever the computer is turned on or woken from sleep, a white LED backlight is forced on, even if I turn it off (`fn+*`) before putting it to sleep.

The issue also appears when leaving the computer idle for a minute or so and then touching the touchpad. It turns on the keyboard backlight.

AaronErhardt, 2023-10-24:
The code is pretty clear. When setting the value to "None", nothing could possibly happen: https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailord/src/led/runtime.rs#L49

> The issue also appears when leaving the computer idle for a minute or so and then touching the touchpad. It turns on the keyboard backlight.

In this case, I assume this is done by the kernel driver. If you prefer your keyboard to always have a certain color (which can be black => no lights btw.), then the "Single" color mode would be the best. If you have another idea to "fix" the overriding of the kernel driver, let me know.

MGlolenstine, 2023-10-24:
I thought `None` meant `no colour on LEDs`. Could that be reworded to make it clearer?
I haven't thought of setting it to `Black` colour 😅, as I stopped at `None`.

I will look around and try to see if there's a way to "fix" the LEDs resetting after waking it up.
Currently, I've just set the brightness level to `0%` using the keyboard shortcuts and it hasn't turned on the LEDs yet.