AaronErhardt/tuxedo-rs #32: Support InfinityBook Pro 14

milgner, 2023-10-31:
I wonder what would need to be done to add support for InfinityBook Pro 14?

In Tailor GUI it currently only says that performance profiles are not available and I can't configure anything.
I'd be very open to adding support - good excuse to write some Rust code :smiley_cat:  - but I'm not sure where to start.

AaronErhardt, 2023-10-31:
If it says that performance profiles are not available, it's very likely that the kernel driver does not support them (though it would be interesting to know whether TCC shows anything). Usually, there shouldn't be any device-specific code required.

So if your InfinityBook Pro works with tuxedo-rs, we could add it to the [list of tested devices](https://github.com/AaronErhardt/tuxedo-rs#tested-hardware).

milgner, 2023-10-31:
TCC shows all the options to configure battery and CPU behaviour. I'm not sure which functionality is part of the performance profiles but currently the Tailor GUI is more or less blank.

AaronErhardt, 2023-10-31:
> the Tailor GUI is more or less blank.

In this case, we have to look into this. Tuxedo-rs supports fan curves, keyboard LED animations and (on some devices) performance profiles (like "performance" or "quite" for example). If those don't work in the GUI, we should dig deeper to figure out why. Additionally, since I haven't touched TCC in a while, I forgot which feature it has, so suggestions about missing features are also great. 

It would be great if you could check your device starting from the lowest abstraction layer: tuxedo-io and tuxedo-keyboard. Those interface with ioctl and sysfs and if they already return something bad, we know where to fix it. You can also find the whole structure of the crates [here](https://github.com/AaronErhardt/tuxedo-rs#structure) :)

jake1909, 2023-10-31:
I am on nixos with infinity book pro 14. I can confirm the blank interface although the controls seems to be there. I can adjust the fans in the blind. Also, the keyboard lightning seems broken. When activating with FN+Space it just blinks once.

nicos68, 2023-11-02:
> Also, the keyboard lightning seems broken. When activating with FN+Space it just blinks once.

Hi,
I just tried on my machine and had the same behavior. I think this is due to the fact that by default tailord (or maybe some other component) sets the led profile to "multiple", which I guess corresponds to rgb keyboard backlight, so if you have a monochrome backlit keyborad as is my case it fails. When I switched my LED profile to "single" then everything would work as before again and I can now adjust the intensity:
[image]

wvengen, 2023-11-19:
```sh
$ sudo dmidecode -s system-product-name
TUXEDO InfinityBook Pro 14 Gen6
$ uname -a
Linux <host> 6.5.0-10005-tuxedo #5 SMP PREEMPT_DYNAMIC <date> x86_64 x86_64 x86_64 GNU/Linux
$ sudo tailor_hwcaps
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "uniwill_wmi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Model ID: "18"
[OK]    Available ODM performance profiles: []
[ERR]   Default ODM performance profile: DevNotAvailable
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [45]
[OK]    Fan speeds [%]: [0]
[OK]    Fan min speed [%]: 20
[INFO]  Webcam control is not available
[OK]    number_of_tdp_devices: 2
[OK]    tdp_descriptors: ["pl1", "pl2"]
[OK]    tdps: [50, 50]
[OK]    max_tdps: [50, 50]
[OK]    min_tdps: [5, 5]
[OK]    Number of LED devices: 0
```

The TUXEDO Control Center 2.0.11 shows: CPU-Temp control, CPU-Frequency control, CPU-Fan control, (Performance) Profiles, Crypt Password, Shutdown Timer, Webcam, Battery charging options (charging profiles, USB-C charging options), Tomte.
Not supported: Keyboard Backlight ("is not (yet) supported by this control interface, or the driver is too old").

SimonBrandner, 2023-12-14:
Is there anything that can be done to aid improving the situation here?

AaronErhardt, 2023-12-14:
> Is there anything that can be done to aid in improving the situation here?

Sure, there are a couple of things to investigate. First of all, it would be nice for me to know what works and what doesn't. Earlier reports say that the GUI is blank, but it seems that tailord is partially working. Someone said the "multiple" color profile caused problems (likely due to monochrome LEDs), but the latest `tailor_hwcaps` report says that there are 0 LED devices detected...

Also if it's true that TCC supports performance profiles whereas their ioctl interface doesn't provide any, that'd also be something to investigate as well...

wvengen, 2023-12-14:
Regarding the keyboard backlight, I can report what happens with the software from Tuxedo:
- Pressing Fn-Space cycles through the states: off, half, full.
- When pressing Fn-Space, an indicator is shown on screen.
- TUXEDO Control Center tells that it "is not (yet) supported by this control interface, or the driver is too old"

It seems that gnome-settings-daemon handles this.
`sudo evtest /dev/input/by-path/platform-tuxedo_keyboard-event` shows, when pressing Fn-Space:
```
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "TUXEDO Keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 56 (KEY_LEFTALT)
    Event code 64 (KEY_F6)
    Event code 125 (KEY_LEFTMETA)
    Event code 191 (KEY_F21)
    Event code 228 (KEY_KBDILLUMTOGGLE)
    Event code 229 (KEY_KBDILLUMDOWN)
    Event code 230 (KEY_KBDILLUMUP)
    Event code 240 (KEY_UNKNOWN)
    Event code 247 (KEY_RFKILL)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1702581466.631438, type 4 (EV_MSC), code 4 (MSC_SCAN), value 3d
Event: time 1702581466.631438, type 1 (EV_KEY), code 228 (KEY_KBDILLUMTOGGLE), value 1
Event: time 1702581466.631438, -------------- SYN_REPORT ------------
Event: time 1702581466.631451, type 1 (EV_KEY), code 228 (KEY_KBDILLUMTOGGLE), value 0
Event: time 1702581466.631451, -------------- SYN_REPORT ------------
```

`dbus-monitor` shows, when pressing Fn-Space:
```
method call time=1702581303.566398 sender=:1.64 -> destination=:1.63 serial=11006 path=/org/gnome/SettingsDaemon/Power; interface=org.gnome.SettingsDaemon.Power.Keyboard; member=Toggle
method return time=1702581303.566543 sender=:1.63 -> destination=:1.64 serial=14007 reply_serial=11006
   int32 0
signal time=1702581303.566696 sender=:1.63 -> destination=(null destination) serial=14008 path=/org/gnome/SettingsDaemon/Power; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.gnome.SettingsDaemon.Power.Keyboard"
   array [
      dict entry(
         string "Brightness"
         variant             int32 0
      )
   ]
   array [
   ]
signal time=1702581303.566743 sender=:1.63 -> destination=(null destination) serial=14009 path=/org/gnome/SettingsDaemon/Power; interface=org.gnome.SettingsDaemon.Power.Keyboard; member=BrightnessChanged
   int32 0
   string "Toggle"
method call time=1702581303.566790 sender=:1.64 -> destination=:1.37 serial=11007 path=/org/gnome/Shell; interface=org.gnome.Shell; member=ShowOSD
   array [
      dict entry(
         string "icon"
         variant             string "keyboard-brightness-symbolic"
      )
      dict entry(
         string "level"
         variant             double 0
      )
   ]
```

When stopping `tccd`, I can confirm that this keeps working.
My conclusion would be that tuxedo-rs doesn't need to touch the keyboard backlight on this device.

wvengen, 2023-12-14:
Looking at changing performance profiles, I tried `strace`-ing `tccd`, which yielded these lines when switching profiles:

```
write(1, "ODMPowerLimitWorker: Set ODM TDP"..., 57) = 57
openat(AT_FDCWD, "/dev/tuxedo_io", O_RDWR) = 21
ioctl(21, _IOC(_IOC_READ, 0xec, 0x5, 0x8), 0x7ffe7baa1bdc) = 0
ioctl(21, _IOC(_IOC_READ, 0xec, 0x6, 0x8), 0x7ffe7baa1bdc) = 0
ioctl(21, _IOC(_IOC_READ, 0xef, 0x1a, 0x8), 0x7ffe7baa1cfc) = 0
ioctl(21, _IOC(_IOC_READ, 0xef, 0x19, 0x8), 0x7ffe7baa1cfc) = 0
ioctl(21, _IOC(_IOC_WRITE, 0xf0, 0x15, 0x8), 0x7ffe7baa1d2c) = 0
ioctl(21, _IOC(_IOC_WRITE, 0xf0, 0x16, 0x8), 0x7ffe7baa1d2c) = 0
close(21)                               = 0
```
(note that there is of course a lot more going on, this was what I could detect being different from no profile switching, but there must be more going on here).
In [this file](https://github.com/tuxedocomputers/tuxedo-control-center/blob/de988b170042d341f5545abcf7e88a2e194b5cd1/src/native-lib/tuxedo_io_lib/tuxedo_io_ioctl.h) we find the ioctl constants, so (hoping I made no mistakes):
* `0xec, 0x05` would be `R_HWCHECK_CL` with `int32_t*` arg (happens all the time)
* `0xec, 0x06` would be `R_HWCHECK_UW` with `int32_t*` arg (happens all the time)
* `0xef, 0x1a` would be `R_UW_TDP2` with `int32_t*` arg
* `0xef, 0x19` would be `R_UW_TDP1` with `int32_t*` arg
* `0xf0, 0x15` would be `W_UW_TDP0` with `int32_t*` arg
* `0xf0, 0x16` would be `W_UW_TDP1` with `int32_t*` arg

Only the first line would differ when a different profile was selected. Syslog showed e.g. when switching to quiet profile:
```
tccd[...]: ODMPowerLimitWorker: Set ODM TDPs ["10 W","15 W","25 W"]
```
and switching back to default profile:
```
tccd[...]: ODMPowerLimitWorker: Set ODM TDPs ["50 W","50 W"]
```

SimonBrandner, 2023-12-26:
For me, the keyboard backlight only flashes and doesn't work with `tuxedo-rs` . The code for setting the brightness (and writing into `/sys/class/leds/white:kbd_backlight/brightness`) works correctly though it seems (according to `lsof` and an experiment with stopping `tailord`) each time a change to `/sys/class/leds/white:kbd_backlight/brightness` is made `tailord` changes the content of the file back to `0`

I'll see if I can figure out anything else

SimonBrandner, 2023-12-26:
Returning from `LedRuntime::run()` before the loop seems to get rid of the issue - seems like I am on a good path

SimonBrandner, 2023-12-26:
So the problem, afaict, seems to be that `ColorProfile::default()` gives a `ColorProfile::Multiple` which isn't supported as the InfinityBook Pro 14 only supports `ControllerMode::Monochrome`

So I assume that we need to adjust the code to take care of the case where the `ControllerMode` is `Monochrome`

Does this match others' understanding of this?

SimonBrandner, 2023-12-26:
My attempt to fix the LED issue is here: https://github.com/AaronErhardt/tuxedo-rs/pull/52

SimonBrandner, 2023-12-30:
What are we missing here? Besides https://github.com/AaronErhardt/tuxedo-rs/issues/46, all seems fine for me?

zoriya, 2023-12-30:
Last time I checked, there were no performance profiles (like stock quiet/normal/performance in tcc).

Also I don't know if this is a tailor feature or a bug on my machine but I have a minimum fan speed limit way higher than the tcc's quiet setting.

jake1909, 2024-01-10:
Can confirm that profiles, led and fan works as expected on my InfinityBook Pro 14 gen 8.
Fantastic!

bobymicroby, 2024-01-29:
@jake1909 Is your nix configuration public so I can take a look? I am on 23.11 and there are no perf profiles, nor anything else.

SimonBrandner, 2024-01-29:
> @jake1909 Is your nix configuration public so I can take a look? I am on 23.11 and there are no perf profiles, nor anything else.

Perf profiles are currently not supported. Otherwise you are probably looking at https://github.com/AaronErhardt/tuxedo-rs/issues/46 which will be fixed by https://github.com/AaronErhardt/tuxedo-rs/pull/60

no1melman, 2024-10-23:
Dunno if I should tack this on here or add a new issue - but found that with the latest xanmod kernel 6.11.3 and enabling tuxedo-rs it fails:

```
rror: builder for '/nix/store/70w1cywfjkrfg6cwgvk1nc5j28zpr39a-tuxedo-keyboard-6.11.3-3.2.14.drv' failed with exit code 2;
       last 25 log lines:
       > Running phase: buildPhase
       > build flags: SHELL=/nix/store/717iy55ncqs0wmhdkwc5fg2vci5wbmq8-bash-5.2p32/bin/bash KDIR=/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/build
       > make -C /nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/build M=/build/source modules
       > make[1]: Entering directory '/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/build'
       >   CC [M]  /build/source/./src/tuxedo_keyboard.o
       > In file included from /build/source/./src/tuxedo_keyboard.c:21:
       > /build/source/./src/tuxedo_keyboard_common.h:64:6: warning: no previous prototype for 'sparse_keymap_report_known_event' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-prototypes-Wmissing-prototypes8;;]
       >    64 | bool sparse_keymap_report_known_event(struct input_dev *dev, unsigned int code,
       >       |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       > In file included from /build/source/./src/tuxedo_keyboard.c:22:
       > /build/source/./src/clevo_keyboard.h:394:19: error: initialization of 'void (*)(struct platform_device *)' from incompatible pointer type 'int (*)(struct platform_device *)' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Werror=incompatible-pointer-types8;;]
       >   394 |         .remove = clevo_keyboard_remove,
       >       |                   ^~~~~~~~~~~~~~~~~~~~~
       > /build/source/./src/clevo_keyboard.h:394:19: note: (near initialization for 'platform_driver_clevo.<anonymous>.remove')
       > In file included from /build/source/./src/tuxedo_keyboard.c:23:
       > /build/source/./src/uniwill_keyboard.h:1259:19: error: initialization of 'void (*)(struct platform_device *)' from incompatible pointer type 'int (*)(struct platform_device *)' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Werror=incompatible-pointer-types8;;]
       >  1259 |         .remove = uniwill_keyboard_remove,
       >       |                   ^~~~~~~~~~~~~~~~~~~~~~~
       > /build/source/./src/uniwill_keyboard.h:1259:19: note: (near initialization for 'platform_driver_uniwill.<anonymous>.remove')
       > cc1: some warnings being treated as errors
       > make[3]: *** [/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/source/scripts/Makefile.build:244: /build/source/./src/tuxedo_keyboard.o] Error 1
       > make[2]: *** [/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/source/Makefile:2077: /build/source] Error 2
       > make[1]: *** [/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/source/Makefile:224: __sub-make] Error 2
       > make[1]: Leaving directory '/nix/store/5vcncc00rk5qvcqzwzgqfsihd7dh9bz4-linux-xanmod-6.11.3-dev/lib/modules/6.11.3-xanmod1/build'
       > make: *** [Makefile:29: all] Error 2
       For full logs, run 'nix log /nix/store/70w1cywfjkrfg6cwgvk1nc5j28zpr39a-tuxedo-keyboard-6.11.3-3.2.14.drv'.
error: 1 dependencies of derivation '/nix/store/1b97zhxbfql279a1xyy7isr25n02rmy6-linux-xanmod-6.11.3-modules.drv' failed to build
error: 1 dependencies of derivation '/nix/store/865jhi53np32lr6wvk3w0f2fl58xfnxv-nixos-system-cl-lt-24-24.11.20241018.4c2fcb0.drv' failed to build
```

am running the same InfinityBook Pro 14 with intel cpu