AaronErhardt/tuxedo-rs #36: Tailord not working until tccd has been started and stopped once

nicos68, 2023-11-03:
I installed tailord from source on my Infinity book S 14 Gen 6 that runs Fedora 38 KDE spin. I deactivated tccd, activated tailord and was successfully able to set the performance profile, with clear impacts on my machine's performance.

Then I rebooted and noticed my machine was slow, and whatever I did the fans were staying quiet. So to make sure I ran rust-analyzer against one of my projects and measured a 50s~1min execution time. I played with the performance profiles on tailord and try all of them but none made any difference on my test's execution time.

Then I disabled and stopped tailord and reactivated and started tccd, and checked the profile in TCC's gui. Then my test's execution time was cut in half and my changes on the performance profile had impacts on my machine's performance again.

Then I disabled and stopped tccd and activated and started tailord again. And then changing the performance profile had impacts on my test's execution time again.

I rebooted my machine and ran these tests a couple of times again and the behavior is consistent, so my belief is that tailord starts working only once tccd has been started and stopped at least once.

AaronErhardt, 2023-11-03:
Thanks for the report! Since I don't have access to an Infinity book, I need your help with debugging this. Tuxedo-rs uses the same kernel interfaces (sysfs and ioctl) as the TCC, so this is definitely strange behavior. Fortunately, due to the modular design of tuxedo-rs it should be quite simple to debug.

I suspect the Infinity book uses hardware from Uniwill (TUXEDO doesn't manufacture everything themselves), but you can find this out rather quickly by using the tuxedo-ioctl crate. You could write a simple test for the performance profile methods (you can use the sudo crate for getting elevated permissions): https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/hal/uniwill.rs#L140-L174

If you start with tccd and tailord disabled and run a testcase that sets a performance profile, we should be able to narrow down the cause.

You might also check whether you're running a recent version of the tuxedo kernel modules.

nicos68, 2023-11-05:
I created this program : 
https://codeberg.org/nise/tuxedo-rs-test

I deactivated both tccd and tailord, rebooted my machine and ran it, which produced this result :
```text
Device id: clevo_acpi
Available profiles: quiet
power_saving
entertainment
performance
```

> You might also check whether you're running a recent version of the tuxedo kernel modules.

Is that what you're talking about ?
```text
$ dnf info tuxedo-keyboard
Last metadata expiration check: 0:00:20 ago on dim. 05 nov. 2023 18:13:13.
Installed Packages
Name         : tuxedo-keyboard
Version      : 3.2.12
Release      : 1
Architecture : noarch
Size         : 1.2 M
Source       : tuxedo-keyboard-3.2.12-1.src.rpm
Repository   : @System
From repo    : copr:copr.fedorainfracloud.org:kallepm:tuxedo-keyboard
Summary      : Kernel module for TUXEDO keyboards
URL          : https://www.tuxedocomputers.com
License      : GPLv3+
Description  : Keyboard & keyboard backlight driver for TUXEDO notebooks
             : meant for DKMS framework.
```

nicos68, 2023-11-06:
> I created this program : https://codeberg.org/nise/tuxedo-rs-test
> 
> I deactivated both tccd and tailord, rebooted my machine and ran it, which produced this result :
I added code to set the preformance profile and did a similar test, it still has no impact on my machine’s performance.

AaronErhardt, 2023-11-07:
> I added code to set the preformance profile and did a similar test, it still has no impact on my machine’s performance.

What happens if you don't use "performance"? Maybe tailord sets the profile to "quiet" or "power_saving", but that should be a quick change in the config.

nicos68, 2023-11-07:
I tried all 4 available profiles, none of them makes any difference.

AaronErhardt, 2023-11-07:
That is really strange. The only thing that tailord does that directly impacts performance is setting the performance profiles. However, since you mentioned that the fans stayed silent, it could be possible that the fan controls didn't work and the CPU had to clock down to prevent overheating. I assume the temperature reporting is probably wrong somehow, which could also be tested through tuxedo-ioctl.

nicos68, 2023-11-18:
Ok so here are my findings :
I tracked down and found the instruction on tccd’s source code that made the difference, and it is [this one](https://github.com/tuxedocomputers/tuxedo-control-center/blob/de988b170042d341f5545abcf7e88a2e194b5cd1/src/service-app/classes/CpuWorker.ts#L179):
`            this.cpuCtrl.setEnergyPerformancePreference('default');`

it turns out that my pc is always set to `power` energy performance preference when it starts up. I don’t know why.

So I added this function to my test programs and run a test procedure:
 1. Boot my PC with both tccd and tailord deactivated
 1. run rust-analyzer, check how long it takes
 1. run my test program
 1. start up tccd
 1. run rust-analyzer again and compare the result.
The tests I ran confirm what I thought, when the performance preference is set correctly and tailord is started _afterwards_, then changing the settings using tailor gui a

closed 2023-11-18 by nicos68