AaronErhardt/tuxedo-rs #40: Support for InfinityBook S 15 Gen6 confirmed

sekret1, 2023-11-16:
I have this notebook and everything works (backlight, fan, performance profile).

Do I have to provide any informations to prove this? If so, what can I provide?

AaronErhardt, 2023-11-16:
You could provide the output of the recently added `tailor_hwcaps` tool (I've just added some information about that [here](https://github.com/AaronErhardt/tuxedo-rs#tested-hardware)), but in general we just trust contributors.

Do you want to send a PR or should I add the device to the list myself?

sekret1, 2023-11-18:
Here's the output:
<pre>
# tailor_hwcaps
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "clevo_acpi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[ERR]   Model ID: NotAvailable
[OK]    Available ODM performance profiles: ["quiet", "power_saving", "entertainment", "performance"]
[OK]    Default ODM performance profile: "performance"
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [44]
[OK]    Fan speeds [%]: [30]
[OK]    Fan min speed [%]: 20
[OK]    Webcam enabled: true
[INFO]  TDP control is not available
[OK]    Number of LED devices: 1
[OK]    LED device number: 0
[OK]    LED device name: "platform:tuxedo_keyboard"
[OK]    LED device function: "kbd_backlight"
[OK]    LED mode: Rgb
[OK]    LED device color: Ok(Color { r: 255, g: 0, b: 0 })
</pre>

I'm wondering why the model ID doesn't show up here… Should I look that one up?

I'm not really all that familiar with providing pull requests etc., so please feel free to add it to the list.

AaronErhardt, 2023-11-18:
> I'm wondering why the model ID doesn't show up here… Should I look that one up?

On clevo hardware it's expected that no model ID shows up, so that's perfectly normal.

I've added the device to the list now, thanks a lot for the info!

closed 2023-11-18 by AaronErhardt