AaronErhardt/tuxedo-rs #41: Broken Fan Control on Gigabyte G5 GD (11th Gen Intel)

enriquezmark36, 2023-11-18:
This laptop had a Clevo chassis and is very well supported by `tccd`. 
As for `tailord`, it has a big problem that the fan control is broken for this device.

By observation (and also the magic of editing the source code to do one thing then abort), it sets the fan speed of the chosen fan just fine but **it also affects the other fan.** 
For example:

Initial Fan Speed: [100, 100]
Set speed of fan 0 to 20: [20, 100]
Set speed of fan 1 to 20: [100, 20]
Set speed of fan 0 to 20: [20, 100]
Set speed of fan 1 to 20: [100, 20]

Anyway, I was inspecting the source code but it appears to be a direct translation from the the original `tccd` code. I did notice something peculiar.

When we read the fan's speed, we always read the same fan instead of the mutable iterator's value. I was expecting this to be using `selected_fan` instead of just `fan`.  https://github.com/AaronErhardt/tuxedo-rs/blob/b80c35e41eb08ff48650221d296f00052d8b95c9/tuxedo_ioctl/src/hal/clevo.rs#L78

This may have an effect of copying the previous fan's speed onto the next fan, or something similar to what I've observed above.

AaronErhardt, 2023-11-18:
Thanks for the detailed report!

I agree, it looks like an oversight caused the error where `fan` instead of `selected_fan` is used in the line that you linked. Since the ioctl interface expects writing all three fan speeds at once, this effects the other fans and causes the behavior you described in the issue. So far, most devices had just one fan, so it makes sense that this wasn't spotted earlier.

Do you want to send a PR or should I take care of the problem?

enriquezmark36, 2023-11-18:
Please take care of it. Thanks.

PR #42 (2023-11-18): fix: Properly set fan speed for multiple fans
Fixes #41
closed 2023-11-18 by AaronErhardt