AaronErhardt/tuxedo-rs #46: Missing elements in GUI

evertvorster, 2023-11-28:
I'm fairly new to this software, but in trying it out there seems to be a pretty bad bug on my system. 

The GUI elements are missing in fan control. 

Here are some screenshots:
[image]

About my system:
```
Operating System: Arch Linux 
KDE Plasma Version: 5.27.9
KDE Frameworks Version: 5.112.0
Qt Version: 5.15.11
Kernel Version: 6.6.2-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 32 × 13th Gen Intel® Core™ i9-13900HX
Memory: 62.5 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: TUXEDO
Product Name: PD5x_7xSNC_SND_SNE
System Version: Not Applicable
```

This is the log in the terminal, hopefully it provides some clue as to why the GUI elements are missing. 
```
[evert@Evert ~]$ sudo systemctl stop tccd.service 
[evert@Evert ~]$ sudo systemctl start tailord
[evert@Evert ~]$ tailor_gui 

(process:27040): Gtk-WARNING **: 09:45:49.283: Unknown key gtk-modules in /home/evert/.config/gtk-4.0/settings.ini
2023-11-28T07:45:49.345503Z  INFO perform: zbus::handshake: new
2023-11-28T07:45:49.345527Z  INFO perform: zbus::handshake: enter
2023-11-28T07:45:49.345557Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: new
2023-11-28T07:45:49.345565Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2023-11-28T07:45:49.345580Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2023-11-28T07:45:49.345585Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: close time.busy=14.3µs time.idle=13.4µs
2023-11-28T07:45:49.345596Z  INFO perform:read_command: zbus::handshake: new
2023-11-28T07:45:49.345599Z  INFO perform:read_command: zbus::handshake: enter
2023-11-28T07:45:49.345603Z  INFO perform:read_command: zbus::handshake: exit
2023-11-28T07:45:49.345607Z  INFO perform: zbus::handshake: exit
2023-11-28T07:45:49.345613Z  INFO perform: zbus::handshake: enter
2023-11-28T07:45:49.345616Z  INFO perform:read_command: zbus::handshake: enter
2023-11-28T07:45:49.345629Z  INFO perform:read_command: zbus::handshake: exit
2023-11-28T07:45:49.345636Z  INFO perform:read_command: zbus::handshake: close time.busy=16.1µs time.idle=23.8µs
2023-11-28T07:45:49.345648Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: new
2023-11-28T07:45:49.345653Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2023-11-28T07:45:49.345662Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2023-11-28T07:45:49.345667Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: close time.busy=9.23µs time.idle=10.2µs
2023-11-28T07:45:49.345676Z  INFO perform:read_command: zbus::handshake: new
2023-11-28T07:45:49.345679Z  INFO perform:read_command: zbus::handshake: enter
2023-11-28T07:45:49.345684Z  INFO perform:read_command: zbus::handshake: exit
2023-11-28T07:45:49.345688Z  INFO perform: zbus::handshake: exit
2023-11-28T07:45:49.345717Z  INFO perform: zbus::handshake: enter
2023-11-28T07:45:49.345721Z  INFO perform:read_command: zbus::handshake: enter
2023-11-28T07:45:49.345729Z  INFO perform:read_command: zbus::handshake: exit
2023-11-28T07:45:49.345732Z  INFO perform:read_command: zbus::handshake: close time.busy=12.0µs time.idle=44.4µs
2023-11-28T07:45:49.345739Z  INFO perform:write_command{command=Begin}: zbus::handshake: new
2023-11-28T07:45:49.345743Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2023-11-28T07:45:49.345750Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2023-11-28T07:45:49.345755Z  INFO perform:write_command{command=Begin}: zbus::handshake: close time.busy=7.26µs time.idle=8.74µs
2023-11-28T07:45:49.345761Z  INFO perform: zbus::handshake: exit
2023-11-28T07:45:49.345766Z  INFO perform: zbus::handshake: close time.busy=201µs time.idle=68.3µs
2023-11-28T07:45:49.345870Z  INFO socket reader: zbus::socket_reader: new
2023-11-28T07:45:49.345881Z  INFO socket reader: zbus::socket_reader: enter
2023-11-28T07:45:49.345890Z  INFO socket reader: zbus::socket_reader: exit
2023-11-28T07:45:49.346190Z  INFO socket reader: zbus::socket_reader: enter
2023-11-28T07:45:49.346303Z  INFO socket reader: zbus::socket_reader: exit
2023-11-28T07:45:49.346984Z  INFO socket reader: zbus::socket_reader: enter
2023-11-28T07:45:49.347054Z  INFO socket reader: zbus::socket_reader: exit
2023-11-28T07:45:49.347607Z  INFO socket reader: zbus::socket_reader: enter
[... 228 lines cut ...]
2023-11-28T07:47:17.604821Z  INFO update_cmd_with_view{cmd_output=() component="tailor_gui::components::factories::profile::Profile" id="0x7ffdfd146e98"}: relm4::factory::sync::builder: enter
2023-11-28T07:47:17.604828Z  INFO update_cmd_with_view{cmd_output=() component="tailor_gui::components::factories::profile::Profile" id="0x7ffdfd146e98"}: relm4::factory::sync::builder: exit
2023-11-28T07:47:17.604832Z  INFO update_cmd_with_view{cmd_output=() component="tailor_gui::components::factories::profile::Profile" id="0x7ffdfd146e98"}: relm4::factory::sync::builder: close time.busy=6.66µs time.idle=19.4µs
2023-11-28T07:47:17.605065Z  INFO socket reader: zbus::socket_reader: enter
2023-11-28T07:47:17.605133Z  INFO socket reader: zbus::socket_reader: exit
2023-11-28T07:47:17.606135Z  INFO socket reader: zbus::socket_reader: enter
2023-11-28T07:47:17.606213Z  INFO socket reader: zbus::socket_reader: exit
2023-11-28T07:47:48.946071Z  INFO update_with_view{input=Add component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: new
2023-11-28T07:47:48.946090Z  INFO update_with_view{input=Add component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: enter
2023-11-28T07:47:48.948658Z  INFO update_with_view{input=Add component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: exit
2023-11-28T07:47:48.948675Z  INFO update_with_view{input=Add component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: close time.busy=2.57ms time.idle=40.3µs
2023-11-28T07:47:51.425980Z  INFO update_with_view{input=Cancel component="tailor_gui::components::new_entry::NewEntryDialog" id="0x7ffdfd146e78"}: relm4::component::sync::builder: new
2023-11-28T07:47:51.426005Z  INFO update_with_view{input=Cancel component="tailor_gui::components::new_entry::NewEntryDialog" id="0x7ffdfd146e78"}: relm4::component::sync::builder: enter
2023-11-28T07:47:51.426015Z  INFO update_with_view{input=Cancel component="tailor_gui::components::new_entry::NewEntryDialog" id="0x7ffdfd146e78"}: relm4::component::sync::builder: exit
2023-11-28T07:47:51.426021Z  INFO update_with_view{input=Cancel component="tailor_gui::components::new_entry::NewEntryDialog" id="0x7ffdfd146e78"}: relm4::component::sync::builder: close time.busy=10.2µs time.idle=34.1µs
2023-11-28T07:47:55.763296Z  INFO update_with_view{input=() component="tailor_gui::modals::about::AboutDialog" id="0x7ffdfd146ec0"}: relm4::component::sync::builder: new
2023-11-28T07:47:55.763316Z  INFO update_with_view{input=() component="tailor_gui::modals::about::AboutDialog" id="0x7ffdfd146ec0"}: relm4::component::sync::builder: enter
2023-11-28T07:47:55.773465Z  INFO update_with_view{input=() component="tailor_gui::modals::about::AboutDialog" id="0x7ffdfd146ec0"}: relm4::component::sync::builder: exit
2023-11-28T07:47:55.773486Z  INFO update_with_view{input=() component="tailor_gui::modals::about::AboutDialog" id="0x7ffdfd146ec0"}: relm4::component::sync::builder: close time.busy=10.1ms time.idle=47.6µs
2023-11-28T07:48:07.410067Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffdfd146ea8"}: relm4::component::sync::builder: new
2023-11-28T07:48:07.410080Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffdfd146ea8"}: relm4::component::sync::builder: enter
2023-11-28T07:48:07.410087Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffdfd146ea8"}: relm4::component::sync::builder: exit
2023-11-28T07:48:07.410090Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7ffdfd146ea8"}: relm4::component::sync::builder: close time.busy=7.11µs time.idle=17.0µs
[evert@Evert ~]$
```

AaronErhardt, 2023-11-28:
The log looks fine. The App loads the profile correctly into memory and should update the graph that shows the profile.

The graph is drawn with cairo and the code tries to get the colors from the active theme so it looks nicely.
However, the API used for receiving the colors was recently deprecated and was more a workaround than a robust solution anyway.
This means that the graph could appear invisible on your device due to foreground and background having the same color.

I've just merged PR [#47](https://github.com/AaronErhardt/tuxedo-rs/pull/47) to introduce a new mechanism for receiving the colors.
Let me know whether this fixed your problem.

evertvorster, 2023-11-29:
Hi there!
I created a tailor-gui-git package in AUR that tracks the master branch, so it's easy to test changes in Arch. 
Unfortunately the fix did not work. I tried with breeze light and breeze dark color schemes to see if that would make a difference. Here is a screenshot of the breeze dark fan control:
[image]

evertvorster, 2023-11-29:
If I click and drag in the window, I can see numbers, so it does look like the interface is working, but on my system you can't see what you are doing... I'll continue to poke at color schemes to see if I can make the graph show up.

AaronErhardt, 2023-11-29:
That's interesting. I thought the new mechanism should be quite robust for selecting colors, but maybe that's not the problem in the first place. You could try changing the color to `RGBA::RED` [here](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailor_gui/src/components/fan_edit.rs#L29) to make sure it's not the color causing the problem.

Otherwise, you might use the inspector (`Ctrl+Shift+D`) to look at the `DrawingArea` and change its properties to find out whether it's properly displayed.

evertvorster, 2023-11-29:
Unfortunately I am not a coder, much less in Rust, so the first suggestion I really don't know how to do. 

With the second suggestion, it's also the first time I tried something like that, and I was unable to find a DrawingArea, much less modify it.

AaronErhardt, 2023-11-29:
> With the second suggestion, it's also the first time I tried something like that, and I was unable to find a DrawingArea, much less modify it.

You could try the following:

1. Open the fan edit dialog
2. Press `Ctrl+Shift+D`
3. Select the top-left button "Select an object"
4. Click on the middle of the dialog (where the DrawingArea is located)
5. The inspector should look like this: [image]
6. Select "Magnifier" in the selector on the left
7. The resulting magnification should look something like that: [image]

Another thing you could try would be to use CSS to change the background color of all widgets.

1. Select "CSS" in the selector on the top
2. Add the following rule
    ```css
    widget {
        background: red;
    }
    ```
    [image]
3. "Unpause" by clicking the pause button on the left
4. Check whether the graph appears (all backgrounds should have turned red)

evertvorster, 2023-11-29:
Hi there!

Thanks for trying to help. 
When selecting magnifier, there is no graph displayed, and when turning the background red, this is what I get:
[image]

evertvorster, 2023-11-29:
I started into X11 mode, and I see some graphical corruption on the tailor-gui... but also the graph!
[image]

On my system, at least, this issue seems related to Wayland. 

No idea why I am getting the graphical corruption either, it's only on the tailor_gui, and the other applications I have tested seem fine.

evertvorster, 2023-11-29:
Just FYI, when running in Wayland mode, I have nvidia_drm.modeset=1 as a kernel parameter. 
In order for X11 to work, I have to disable that.

evertvorster, 2023-11-29:
Adding a bit more info here, I have fractional scaling of 120% on both of my monitors.

SimonBrandner, 2023-12-29:
I am wondering if this could be an issue with Relm/GTK... :thinking:

AaronErhardt, 2023-12-29:
@SimonBrandner I assume it's a GTK issue because Relm4 does not add any drawing code and just uses GTK widgets. Do you also have the issue?

SimonBrandner, 2023-12-29:
Yes, I do have it (assuming since I am on Wayland - do we have anyone not seeing this on Wayland/seeing this on X11?) - I temporarily solved it by doing `GDK_BACKEND=x11 tailor_gui`

AaronErhardt, 2023-12-29:
It works for me on Wayland without problems (just rebuilt everything with the release profile to make it sure it still works). It used to work as native installation as well on Fedora. Maybe NixOS has a different GTK version? You could also try to install it as flatpak from source (using GNOME Builder or Flatpak VSCode plugin).

SimonBrandner, 2023-12-29:
So the weird thing is that when running the locally built version, it defaults to `x11`, so the issue is not present, but when I run it with `GDK_BACKEND=wayland ./result/bin/tailor_gui`, the issue is present

How are you starting it?

AaronErhardt, 2023-12-29:
I just run it without any relevant environment variables (you can read the config from the flatpak manifest) so it defaults to wayland. I wonder why it defaults to X11 on your system when GTK prefers wayland for quite some time now AFAIK. What are other GTK apps choosing on your system?

SimonBrandner, 2023-12-30:
I'd like to test the behavior of Flatpak on my system but I have trouble getting it to run... Do you perhaps have an idea of what's going on? There doesn't seem to be any `org.freedesktop.Sdk.Extension.rust-stable/x86_64/45` flatpak package...
 
```
❯ flatpak install org.gnome.Sdk//45 org.freedesktop.Sdk.Extension.rust-stable//22.08 org.gnome.Platform//45 runtime/org.freedesktop.Sdk.Extension.llvm15//22.08
flatpak-builder --user flatpak_app tailor_gui/build-aux/com.github.aaronerhardt.Tailor.json

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/simon/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
Skipping: org.gnome.Sdk/x86_64/45 is already installed
Skipping: org.freedesktop.Sdk.Extension.rust-stable/x86_64/22.08 is already installed
Skipping: org.gnome.Platform/x86_64/45 is already installed
Skipping: org.freedesktop.Sdk.Extension.llvm15/x86_64/22.08 is already installed
Downloading sources
Initializing build dir
error: Requested extension org.freedesktop.Sdk.Extension.rust-stable/x86_64/45 not installed
Error: Child process exited with code 1
```

PS: Have you perhaps considered making a Discord Server/Matrix Room or Space for tuxedo-rs? I think it might be a little better for real-time discussions than GH...

AaronErhardt, 2023-12-30:
https://matrix.to/#/%23tuxedo-rs:matrix.org

AaronErhardt, 2023-12-30:
In my experience, flatpak-builder is not the greatest tool to build flatpaks. GNOME Builder and VSCode with the Flatpak extension work like a charm though, you basically just need to click a button and it builds and runs the app.

I mean, I get one step further with flatpak-builder, but then it complains that Cargo.toml is missing...

jake1909, 2024-01-19:
Running GDK_SCALE=1 tailor_gui solves this for me.

AaronErhardt, 2024-01-26:
Thanks for the hint. It actually seems to be related to monitoring scaling and I was able to replicate it (I only use 100% usually, so it never happened to me before). Once I find the cause, I'll try to come up with a patch.

AaronErhardt, 2024-01-26:
I've now identified and fixed the cause. The bug was actually caused by the scaling of the window. One part of the code took the scaling into account and doubled the size of the canvas where the plot is drawn, but also added scaling to the canvas, so that the actual coordinates stayed identical. Another part of the code was not aware of the scaling and thought it needed twice as large coordinates. The code therefore created coordinates twice as large as they should have been, mostly drawing outside of the visible canvas.

You can already try out the fix in #60. I will merge the PR in the coming days and issue a new release of tuxedo-rs soon as well :)

closed 2024-02-08 by AaronErhardt

evertvorster, 2024-03-18:
Hi there!

I just tried version 0.2.4 of the tailor gui. I can confirm that with no scaling I now see the graph in fan control. 

Unfortunately, if I scale my monitors the way I like them, the graph again does not show up. 

I am running Wayland and KDE, and have one monitor scaled to 150% and the other to 120% (One monitor is QHD, and the other 4K)

However, it also does not show up when I set the two monitors to the same scaling, only when scaling is completely disabled. 

If there is a newer version than 0.2.4 in the works, please feel free to close this report again.

PR #60 (2024-02-08): WIP: feat(gui)!: Update Relm4 dependency
Update Relm4 to pull in a recent fix for custom drawing what resulted in missing elements.

Fixes #46 

I will merge this once the drawing fix has been released with Relm4 on crates.io so we don't need the git dependencies anymore.