AaronErhardt/tuxedo-rs #51: Advanced Fan control

MGlolenstine, 2023-12-26:
I'm using tuxedo-rs instead of the TCC on my NixOS machine.

Fan seems to be causing problems and spinning up too soon and being loud when opening up tabs in the browser, let alone compiling code.

I had no such issues with TCC and I'm wondering if there's a way to disable the "fan-curve" block, which is understandably there for the CPU protection, but prevents me from setting my laptop to "quiet" mode.

I have set my fan profile manually to
```
[{"temp":25,"fan":0},{"temp":30,"fan":10},{"temp":42,"fan":10},{"temp":58,"fan":16},{"temp":61,"fan":22},{"temp":70,"fan":25},{"temp":80,"fan":100}]
```
but the daemon complains about needing to have 40% at 70C.

There could be another issue with the CPU reporting temperature (or the tuxedo-rs interpreting it wrong) as well.

My model isn't one of the `Tuxedo` ones, but it's a [Clevo one](https://laptopwithlinux.com/product/clevo-nl50nu-amd-laptop/).
This far I've had no issues with TCC or it overheating. I'd love to help and make it supported. What would be the first steps?

AaronErhardt, 2023-12-27:
> but the daemon complains about needing to have 40% at 70C.

I've added a minimum fan speed of `T-50 * 2` starting from 50°C to prevent hardware damage due to careless usage. Currently, there's no option for disabling this limitation, but we could add a switch for disabling the safety limits for advanced usage.

MGlolenstine, 2024-01-04:
I have found the culprit that has caused the fan to become loud and start spinning fast for no reason.
At first, I thought the issue was the driver, as it started when I switched from TCC to tuxedo-rs, but that seems to not be the case.

While the TCC did have lower RPMs at higher temperatures, the culprit seems to be the lack of liquid metal present on the CPU die.
I opened up my laptop to blow the dust, but I was surprised when it was immaculate on the inside. After I removed the heat sink for the CPU. I've noticed that all of the liquid metal has solidified and there was next to no heat transfer.
I've cleaned it up and reapplied normal thermal paste and it's been working like a bliss ever since.

In the end, the solution was hardware-related.