AaronErhardt/tuxedo-rs #56: Aura Gen 1 only has 2 governors: performance + shedutil

LoveIsGrief, 2023-12-30:
Hi,

I'm on NixOS

 - system: `"x86_64-linux"`
 - host os: `Linux 6.1.68, NixOS, 23.11 (Tapir), 23.11.2217.d02d818f22c7`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.18.1`
 - channels(root): `"nixos-23.11"`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos`

And have tried setting my CPU to quiet, powersave and other modes, but the fan just keeps going. So I checked and the CPU governor is set to `performance`

```
$ cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
performance
performance
performance
performance
performance
performance
performance
performance
```

There no other governors available

```
$ cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_available_governors
performance schedutil 
performance schedutil 
performance schedutil 
performance schedutil 
performance schedutil 
performance schedutil 
performance schedutil 
performance schedutil 
```

I don't know if this is a NixOS issue or a tuxedo-rs issue, but my battery definitely doesn't  last long anymore. 3h tops even for light work.

AaronErhardt, 2023-12-30:
Tuxedo-rs offers so called "ODM performance profiles" similar to TCC. They are offered by the TUXEDO driver via ioctl, but I couldn't find any into about what they do exactly myself. I assume they set a certain governor or add some clock frequency limits.

I also have an Aura 15 Gen 1 (with Fedora 39) and for me, the performance profiles seem to work as intended. I can check the sysfs paths you tried later to see whether I also can only see one governor and whether the performance profiles affect them.

AaronErhardt, 2023-12-30:
I have tested the sysfs paths on my system now and have the following results:

I have a full set of available governors on all cores (conservative ondemand userspace powersave performance scheduti) and by default scheduti is enabled. The "ODM performance profiles" I can set via tuxedo-rs make no difference regarding this. I couldn't really see a difference regarding CPU frequencies either when setting the performance profile from "quiet" to "performance" for example, but I think my laptop stays quieter with "power_safe" or "quiet" somehow (maybe it is easier to detect with connected power supply, I'm just on battery currently).

LoveIsGrief, 2023-12-30:
I added https://github.com/NixOS/nixos-hardware/blob/22ae59fec26591ef72ce4ccb5538c42c5f090fe3/common/cpu/amd/pstate.nix#L26 to my nixos config and now have this in my kernel boot commandline `amd_pstate=passive`.

Got it from `journalctl -b0 --grep "Command line" |& grep -Eo 'initrd.*'`

Hmm... my scaling_driver should be something AMD, but instead I get

```sh
$ cat /sys/devices/system/cpu/cpu1/cpufreq/scaling_driver
acpi-cpufreq
$ cat /sys/devices/system/cpu/cpu1/cpufreq/scaling_available_governors 
performance schedutil 
```

These are my kernel modules

```sh
$ lsmod | grep -oE '^(\w+)?amd\w+'   
snd_sof_amd_rembrandt
snd_sof_amd_renoir
snd_sof_amd_acp
amdgpu
```

and this is my kernel

```sh
$ uname -sr
Linux 6.1.69
```

What does the output of those look like for you?

I'll see if I can set the governor to schedutil too. Thanks for checking on your laptop.