AaronErhardt/tuxedo-rs #59: Support for Sirius 16 gen 1

Szybet, 2024-01-24:
Hello
Sadly I must report that tailor_hwcaps on arch linux ( using all the aur packages ) on the Sirius 16 gen 1 reports:
```
[FATAL] Connecting to ioctl interface failed: DevNotAvailable
[OK]    Number of LED devices: 0
```
The same in the GUI
I'm willing to provide additional info and try out other things to make it work

AaronErhardt, 2024-01-26:
Do you have a recent version of the tuxedo drivers installed? What does `modinfo tuxedo_io` return on your machine?

Szybet, 2024-01-26:
latest ones from aur
```
» modinfo tuxedo_io
filename:       /lib/modules/6.6.13-1-lts/updates/dkms/tuxedo_io.ko.zst
alias:          wmi:ABBC0F6F-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6E-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          acpi*:CLV0001:*
alias:          wmi:ABBC0F6D-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F6B-8EA1-11D1-00A0-C90629100000
license:        GPL
version:        0.3.9
author:         TUXEDO Computers GmbH <tux@tuxedocomputers.com>
description:    Hardware interface for TUXEDO laptops
srcversion:     C61A4D6C57A1721F14A37E6
depends:        tuxedo_keyboard
retpoline:      Y
name:           tuxedo_io
vermagic:       6.6.13-1-lts SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        4D:9D:FF:C0:C3:6F:24:E4:4A:C1:15:81:24:55:CF:83:04:1E:1B:AB
sig_hashalgo:   sha512
signature:      66:07:8A:E0:0C:78:13:28:86:75:73:85:99:16:68:98:D6:7D:12:40:
		6E:91:9E:14:18:7A:26:C4:2B:13:93:53:E8:7D:43:DA:DC:4D:AA:0B:
		76:21:01:76:F9:7E:A2:ED:F5:61:11:20:36:5D:BD:BC:37:FD:79:E6:
		20:CF:B6:18:B1:B0:4F:C7:50:4B:B5:2A:96:91:16:E3:33:6B:D3:6A:
		FC:D9:B7:35:6E:B9:06:E6:D7:2C:07:70:5A:DB:A0:F3:CF:15:BE:6F:
		DF:00:E3:F3:E6:B3:98:0E:C1:70:05:00:E9:03:F2:16:10:1E:1C:6F:
		90:7F:0C:9B:D9:74:AC:31:B7:F3:3F:26:23:0B:58:B9:0A:34:39:72:
		6D:41:FD:20:92:16:38:32:57:0E:BF:0B:71:A3:E8:B6:FC:F8:D3:3E:
		DF:FB:D7:60:D7:FC:6C:9B:E3:A2:FD:5E:6E:57:02:01:80:FA:3D:22:
		0E:A6:79:12:D0:65:7D:CD:1B:2D:40:EA:76:CB:86:06:45:30:C4:11:
		88:B9:F5:B9:2F:E8:B7:58:88:AF:33:8C:6E:E4:96:70:02:B5:FF:CA:
		3A:3C:EF:21:45:64:2F:B6:37:2E:1D:88:6D:B3:8F:D5:1F:A9:BB:1B:
		4A:04:D9:A7:BF:0C:0B:4F:86:89:F5:D1:E2:9B:2C:F1
```

AaronErhardt, 2024-01-26:
Ok, you have the same version as I do, so that shouldn't be the problem. However, tuxedo-rs still complains...

You could check, whether the ioctl file descriptor exists as a next step:

```sh
ls /dev | grep tuxedo
```

Expected output:

```
crw-------@  234,0 root  26 Jan 21:13 tuxedo_io
```

I have the suspicion though, that it's an error from here: https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/hal/mod.rs#L49

If that's true, it would mean that you have hardware from neither uniwill or clevo, which have historically been the only manufacturers that produced hardware for TUXEDO...

Szybet, 2024-01-26:
```
» ls -shal /dev | grep tuxedo
   0 crw-------   1 root   root   234,     0 01-26 20:15 tuxedo_io
```
you could supply me a binary that gives you an anwer for that

AaronErhardt, 2024-01-26:
The problem is that it is not straight forward to figure this out.

Have you already worked with Rust?
If yes, the easiest approach would be to run the tests [here](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/read.rs#L133) and slightly modify them until we found the root cause.

Szybet, 2024-01-26:
Yep I had
So do I just add a debug information if `Err(IoctlError::DevNotAvailable)` or just run the tests? If tests, do I need to run them as root? do I need some other clevo modules?

AaronErhardt, 2024-01-26:
You can just run both tests (`test_cl_read` and `test_uw_read`) and first see whether they complete. You don't need root permissions up front, the tests will automatically try to get the required permissions (and might ask for your password).

Szybet, 2024-01-26:
result here:
```
running 2 tests
thread 'read::test::test_cl_read' panicked at tuxedo_ioctl/src/read.rs:149:13:
assertion failed: uw::hw_check(&file).unwrap() == 1
stack backtrace:
thread 'read::test::test_uw_read' panicked at tuxedo_ioctl/src/read.rs:169:13:
assertion failed: cl::hw_check(&file).unwrap() == 1

running 2 tests
               thread 'read::test::test_uw_read' panicked at tuxedo_ioctl/src/read.rs:169:13:
                                                                                             assertion failed: cl::hw_check(&file).unwrap() == 1
                                                                                                                                                stack backtrace:
                                                                                                                                                                thread 'read::test::test_cl_read' panicked at tuxedo_ioctl/src/read.rs:149:13:
                                                                assertion failed: uw::hw_check(&file).unwrap() == 1
                                                                                                                      0:     0x579b0cd6c88c - std::backtrace_rs::backtrace::libunwind::trace::ha637c64ce894333a
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
   1:     0x579b0cd6c88c - std::backtrace_rs::backtrace::trace_unsynchronized::h47f62dea28e0c88d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x579b0cd6c88c - std::sys_common::backtrace::_print_fmt::h9eef0abe20ede486
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x579b0cd6c88c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hed7f999df88cc644
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:44:22
   4:     0x579b0cd970f0 - core::fmt::rt::Argument::fmt::h1539a9308b8d058d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/rt.rs:142:9
   5:     0x579b0cd970f0 - core::fmt::write::h3a39390d8560d9c9
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/mod.rs:1120:17
   6:     0x579b0cd6a17f - std::io::Write::write_fmt::h5fc9997dfe05f882
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/io/mod.rs:1762:15
   7:     0x579b0cd6c674 - std::sys_common::backtrace::_print::h894006fb5c6f3d45
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:47:5
   8:     0x579b0cd6c674 - std::sys_common::backtrace::print::h23a2d212c6fff936
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:34:9
   9:     0x579b0cd6e337 - std::panicking::default_hook::{{closure}}::h8a1d2ee00185001a
  10:     0x579b0cd6e09f - std::panicking::default_hook::h6038f2eba384e475
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:292:9
  11:     0x579b0cd6e7b8 - std::panicking::rust_panic_with_hook::h2b5517d590cab22e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:779:13
  12:     0x579b0cd6e669 - std::panicking::begin_panic_handler::{{closure}}::h233112c06e0ef43e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:649:13
  13:     0x579b0cd6cd56 - std::sys_common::backtrace::__rust_end_short_backtrace::h6e893f24d7ebbff8
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:170:18
  14:     0x579b0cd6e402 - rust_begin_unwind
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:645:5
  15:     0x579b0cccdd95 - core::panicking::panic_fmt::hbf0e066aabfa482c
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
  16:     0x579b0cccde33 - core::panicking::panic::h195fc2a96298d4c3
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:127:5
  17:     0x579b0cce6909 - tuxedo_ioctl::read::test::test_cl_read::he1be9663b7dc550f
                               at /mnt/data/projects/other/tuxedo-rs/tuxedo_ioctl/src/read.rs:149:13
  18:     0x579b0cce9517 - tuxedo_ioctl::read::test::test_cl_read::{{closure}}::h452331c9068a447b
                               at /mnt/data/projects/other/tuxedo-rs/tuxedo_ioctl/src/read.rs:133:22
  19:     0x579b0ccd36e6 - core::ops::function::FnOnce::call_once::hea53add40e22a61f
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/ops/function.rs:250:5
  20:     0x579b0cd3114f - core::ops::function::FnOnce::call_once::h2b5d431d5046312d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/ops/function.rs:250:5
  21:     0x579b0cd3114f - test::__rust_begin_short_backtrace::h7714669af176e207
[... cut ...]
test read::test::test_cl_read ... FAILED

failures:

failures:
    read::test::test_cl_read
    read::test::test_uw_read

test result: FAILED. 0 passed; 2 failed; 0 ignored; 0 measured; 1 filtered out; finished in 0.01s

error: test failed, to rerun pass `-p tuxedo_ioctl --lib`

 *  The terminal process "cargo 'test', '--package', 'tuxedo_ioctl', '--lib', '--', 'read::test', '--nocapture'" terminated with exit code: 101. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

Looks bad

AaronErhardt, 2024-01-26:
Ok, so the test results actually proved my first suspicion. Both tests have an if statement to check for a certain manufacturer and since we just had two so far, they assert that if you don't have clevo hardware, that you must have uniwill hardware instead and vice-versa. And since your manufacturer is none of both, the test cases fail on this assertions.

I guess we have to look at the code of TCC and check whether new heuristics were added there recently to account for a potential third manufacturer. Give me a couple of minutes.

Szybet, 2024-01-26:
I looked arround tuxedo drivers repo
```
sirius :: projects/other/tuxedo-drivers 1 » grep -R Sirius
debian/changelog:  * Add nb04 driver collection (so far used by Sirius 16 Gen1)
src/tuxedo_nb04/tuxedo_nb04_power_profiles.c:	// Sirius uses other platform control interface
tuxedo-drivers.spec.in:- Add nb04 driver collection (so far used by Sirius 16 Gen1)
```

the code name for the sirius 16 gen 1 is nb04, apart from that I didn't found in the code anything that could point me who manufactured this device.

this line is interesting:
```
Sirius uses other platform control interface
```

AaronErhardt, 2024-01-26:
I think you're right. For whatever reason, TUXEDO is not able to come up with a driver that has a consistent interface, but instead they hack together a solution for each vendor and call it a day... Plus they don't document anything.

Anyway, I found a piece of code that indicates that the features commonly implemented via ioctl are exposed via sysfs in recent devices: https://github.com/tuxedocomputers/tuxedo-control-center/blob/master/src/service-app/classes/ODMProfileWorker.ts#L30-L42

If I'm correct `sudo cat /sys/bus/platform/devices/tuxedo_platform_profile/platform_profile` should return something for you.

Szybet, 2024-01-26:
Nope
```
sirius :: / 130 » sudo find /sys | grep platform_profile
/sys/kernel/btf/platform_profile
/sys/kernel/debug/printk/index/platform_profile
/sys/firmware/acpi/platform_profile
/sys/firmware/acpi/platform_profile_choices
/sys/module/platform_profile
/sys/module/platform_profile/initsize
/sys/module/platform_profile/uevent
/sys/module/platform_profile/notes
/sys/module/platform_profile/notes/.note.Linux
/sys/module/platform_profile/notes/.note.gnu.build-id
/sys/module/platform_profile/notes/.note.gnu.property
/sys/module/platform_profile/taint
/sys/module/platform_profile/srcversion
/sys/module/platform_profile/holders
/sys/module/platform_profile/holders/amd_pmf
/sys/module/platform_profile/refcnt
/sys/module/platform_profile/coresize
/sys/module/platform_profile/initstate
/sys/module/platform_profile/sections
/sys/module/platform_profile/sections/__patchable_function_entries
/sys/module/platform_profile/sections/.orc_unwind
/sys/module/platform_profile/sections/.note.Linux
/sys/module/platform_profile/sections/.strtab
/sys/module/platform_profile/sections/__mcount_loc
/sys/module/platform_profile/sections/.bss
/sys/module/platform_profile/sections/.orc_unwind_ip
/sys/module/platform_profile/sections/.return_sites
/sys/module/platform_profile/sections/.gnu.linkonce.this_module
/sys/module/platform_profile/sections/.symtab
/sys/module/platform_profile/sections/.rodata
/sys/module/platform_profile/sections/.note.gnu.build-id
/sys/module/platform_profile/sections/.text
/sys/module/platform_profile/sections/.call_sites
/sys/module/platform_profile/sections/.data
/sys/module/platform_profile/sections/__bug_table
/sys/module/platform_profile/sections/.rodata.str1.1
/sys/module/platform_profile/sections/.note.gnu.property
/sys/module/platform_profile/sections/.retpoline_sites
/sys/module/platform_profile/sections/__ksymtab_strings
/sys/module/platform_profile/sections/.orc_header
/sys/module/platform_profile/sections/__ksymtab_gpl
/sys/module/platform_profile/sections/.rodata.str1.8
/sys/module/platform_profile/sections/.export_symbol
```
```
sirius :: / 1 » ls /sys/bus/platform/devices/
ACPI0003:00    acp_ps_pdm_dma.0   AMDI0010:03  AMDI0102:00   i8042             microcode    PNP0103:00  PNP0C0A:00  PNP0C14:00  regulatory.0  SMUX1234:00    TFAN0000:00
acpi-cpufreq   alarmtimer.0.auto  AMDI0030:00  dmic-codec.0  INT33D3:00        MSFT0101:00  PNP0800:00  PNP0C0C:00  PNP0C14:01  rtc-efi.0     snd-soc-dummy  TUXI0000:00
acp_ps_mach.0  AMDI0009:00        AMDI0052:00  efivars.0     intel_rapl_msr.0  pcspkr       PNP0C09:00  PNP0C0D:00  reg-dummy   serial8250    sp5100-tco
```

Szybet, 2024-01-26:
I don't have tuxedo control center installed, but I don't think it should matter?

AaronErhardt, 2024-01-26:
Wait, but you have `/sys/firmware/acpi/platform_profile`. I've overlooked that first, but that seems to be the second path exposed by the driver (don't ask me why there are two paths). Can you test that?

Szybet, 2024-01-26:
Why would they add it to acpi...
```
sirius :: ~ » cat /sys/firmware/acpi/platform_profile
balanced
```
I also don't have the tuxedo patched kernel, but I don't think it's needed?

AaronErhardt, 2024-01-26:
> Why would they add it to acpi...

Maybe they haven't. I could imagine that they just adapted an existing driver for similar hardware. Their driver code doesn't look pretty and has zero comments so it's hard to guess what they are even doing. I don't think that path would be exposed by sysfs without the driver though.

For tuxedo-rs this means, we need to add another abstraction layer... Essentially we need a sysfs (or maybe two) abstraction on top of the two existing ioctl interfaces. Alternatively, we could write a better driver, which overall would be the better solution. I don't have time for such things though...

Szybet, 2024-01-26:
Me neither for now
Thanks for your time

Szybet, 2024-02-02:
The manufacturer is Schenker:
https://www.techpowerup.com/317582/schenker-built-tuxedo-sirius-16-notebook-powered-by-ryzen-7040-showcased
https://www.schenker-tech.de/en/schenker-laptops/

Szybet, 2024-08-22:
The keyboard firmware has those strings:
```
Emdoor-AG958-02_04_29
ITE Tech. Inc.
```
quick google search: 

http://www.emdoordigi.com/company_new/74.html

https://www.ite.com.tw/tw

einsiedlerspiel, 2025-05-09:
For future reference: I'm running into this same issue with the Tuxedo Pulse 14 Gen4.