AaronErhardt/tuxedo-rs #66: tailord fails to run

Kuchenmampfer, 2024-03-15:
I just installed tailord and tailor-gui from the AUR on my tuxedo aura 15 gen1 laptop. Sadly, tailord fails to run with the following output:
```
INFO start_runtime: tailord: Starting tailord
 WARN start_runtime: tailord::profiles: Failed to load active profile at `/etc/tailord/active_profile.json`: IOError("No such file or directory (os error 2)")
 INFO start_runtime: tailord::profiles: Loaded profile at `/etc/tailord/active_profile.json`: ProfileInfo { fans: ["default"], leds: [], performance_profile: None }
ERROR start_runtime: tailord::profiles: Failed to load fan color profile called `default`: `org.freedesktop.DBus.Error.IOError: No such file or directory (os error 2)`
 WARN start_runtime: tailord: No tuxedo ioctl interface available: No such file or directory (os error 2)
thread 'main' panicked at tailord/src/main.rs:164:10:
called `Result::unwrap()` on an `Err` value: FDO(AccessDenied("Request to own name refused by policy"))
```
I couldn't find more about fan color profiles or ioctl (except that its a c library), so I am reporting this here and hope someone can help.

Thanks in advance :)

AaronErhardt, 2024-03-15:
The profiles and ioctl warnings are not ideal, but they're not the cause of the crash. It seems that the service failed to register the DBUS service.

Actually, I don't know who created the AUR package and whether it sets the files up correctly. Also, maybe you're not using systemd? You could also try installing it manually to figure out whether it's a packaging issue. The instructions can be found [here](https://github.com/AaronErhardt/tuxedo-rs?tab=readme-ov-file#tailord).

Kuchenmampfer, 2024-03-15:
Ah okay. I am using systemd, but yeah, maybe the packaging is wrong. I will try to install it manually 👍

Kuchenmampfer, 2024-03-15:
Alright, now it starts, but does not find the fan. 
[image]
[image]
The keybooard now constantly changes color and reacts to its shortcuts, but not to tailor-gui which I build from your instructions without flatpak-builder because there, some version numbers seem to be messed up.

Kuchenmampfer, 2024-03-16:
Now it seems to work, maybe a reboot was needed.

closed 2024-03-16 by Kuchenmampfer