AaronErhardt/tuxedo-rs #68: No connection from git commit e731aba

evertvorster, 2024-03-19:
Hi there!

It took a little while for me to do, but now I can check various git updates as they were made. 
Unfortunately, there seems to be an issue in this system. 

When tailor_gui is launched, I get this error:
[image]

Going through the various git commits, I can see the last one that worked properly was 74b863e
b597412 does not build for me...
e731aba is the first git commit that shows the error. 

This error is also apparent on the latest git commit, fcca331

More about my system:
```
Operating System: Arch Linux 
KDE Plasma Version: 6.0.2
KDE Frameworks Version: 6.0.0
Qt Version: 6.6.2
Kernel Version: 6.6.22-1-lts (64-bit)
Graphics Platform: Wayland
Processors: 32 × 13th Gen Intel® Core™ i9-13900HX
Memory: 62.5 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: TUXEDO
Product Name: PD5x_7xSNC_SND_SNE
```

AaronErhardt, 2024-03-25:
I think you're running an outdated version of tailord. I can't reproduce the issue on my system.

evertvorster, 2024-03-26:
Hi there!

Thanks for looking into this. 
I am running the latest release of tailord, 0.2.4
I actually went one step further, and checked out the code from master, and built 0.2.4 from there, and still I have this issue.

evertvorster, 2024-03-26:
Here is some command line output:
```
[evert@Evert tailor-gui]$ GDK_BACKEND=x11 tailor_gui

(process:649139): Gtk-WARNING **: 07:45:20.903: Unknown key gtk-modules in /home/evert/.config/gtk-4.0/settings.ini
2024-03-26T05:45:21.013152Z  INFO perform: zbus::handshake: new
2024-03-26T05:45:21.013173Z  INFO perform: zbus::handshake: enter
2024-03-26T05:45:21.013222Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: new
2024-03-26T05:45:21.013232Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2024-03-26T05:45:21.013239Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2024-03-26T05:45:21.013242Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2024-03-26T05:45:21.013244Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2024-03-26T05:45:21.013247Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: close time.busy=9.61µs time.idle=15.6µs
2024-03-26T05:45:21.013253Z  INFO perform:read_command: zbus::handshake: new
2024-03-26T05:45:21.013256Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013259Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013262Z  INFO perform: zbus::handshake: exit
2024-03-26T05:45:21.013386Z  INFO perform: zbus::handshake: enter
2024-03-26T05:45:21.013402Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013416Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013419Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013421Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013424Z  INFO perform:read_command: zbus::handshake: close time.busy=20.2µs time.idle=150µs
2024-03-26T05:45:21.013431Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: new
2024-03-26T05:45:21.013434Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2024-03-26T05:45:21.013439Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2024-03-26T05:45:21.013441Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2024-03-26T05:45:21.013444Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2024-03-26T05:45:21.013446Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: close time.busy=7.49µs time.idle=8.02µs
2024-03-26T05:45:21.013450Z  INFO perform:read_command: zbus::handshake: new
2024-03-26T05:45:21.013452Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013456Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013458Z  INFO perform: zbus::handshake: exit
2024-03-26T05:45:21.013475Z  INFO perform: zbus::handshake: enter
2024-03-26T05:45:21.013477Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013481Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013483Z  INFO perform:read_command: zbus::handshake: enter
2024-03-26T05:45:21.013485Z  INFO perform:read_command: zbus::handshake: exit
2024-03-26T05:45:21.013489Z  INFO perform:read_command: zbus::handshake: close time.busy=9.87µs time.idle=29.6µs
2024-03-26T05:45:21.013493Z  INFO perform:write_command{command=Begin}: zbus::handshake: new
2024-03-26T05:45:21.013495Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2024-03-26T05:45:21.013499Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2024-03-26T05:45:21.013502Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2024-03-26T05:45:21.013504Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2024-03-26T05:45:21.013506Z  INFO perform:write_command{command=Begin}: zbus::handshake: close time.busy=6.62µs time.idle=7.09µs
2024-03-26T05:45:21.013510Z  INFO perform: zbus::handshake: exit
2024-03-26T05:45:21.013512Z  INFO perform: zbus::handshake: enter
2024-03-26T05:45:21.013514Z  INFO perform: zbus::handshake: exit
2024-03-26T05:45:21.013516Z  INFO perform: zbus::handshake: close time.busy=199µs time.idle=170µs
2024-03-26T05:45:21.013592Z  INFO socket reader: zbus::socket_reader: new
2024-03-26T05:45:21.013594Z  INFO socket reader: zbus::socket_reader: enter
2024-03-26T05:45:21.013599Z  INFO socket reader: zbus::socket_reader: exit
2024-03-26T05:45:21.013637Z  INFO socket reader: zbus::socket_reader: enter
2024-03-26T05:45:21.013694Z  INFO socket reader: zbus::socket_reader: exit
2024-03-26T05:45:21.014091Z  INFO socket reader: zbus::socket_reader: enter
2024-03-26T05:45:21.014131Z  INFO socket reader: zbus::socket_reader: exit
2024-03-26T05:45:21.014654Z  INFO socket reader: zbus::socket_reader: enter
2024-03-26T05:45:21.014699Z  INFO socket reader: zbus::socket_reader: exit
2024-03-26T05:45:21.014723Z  INFO socket reader: zbus::socket_reader: enter
2024-03-26T05:45:21.014738Z  INFO socket reader: zbus::socket_reader: exit
2024-03-26T05:45:21.014740Z  INFO socket reader: zbus::socket_reader: close time.busy=169µs time.idle=980µs
2024-03-26T05:45:21.025728Z  INFO update_cmd_with_view{cmd_output=SetInitializedState { error: Some("Serialization error: `missing field `mode` at line 1 column 70`") } component="tailor_gui::app::App" id="0x7fff89b339e8"}: relm4::component::sync::builde [... cut ...]
2024-03-26T05:45:27.597050Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7fff89b339e8"}: relm4::component::sync::builder: enter
2024-03-26T05:45:27.597063Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7fff89b339e8"}: relm4::component::sync::builder: exit
2024-03-26T05:45:27.597067Z  INFO update_with_view{input=Quit component="tailor_gui::app::App" id="0x7fff89b339e8"}: relm4::component::sync::builder: close time.busy=13.2µs time.idle=22.1µs
```

dimitarkolevdtd, 2024-04-01:
I have the same issue on a fresh install of Arch Linux on the TUXEDO Stellaris, using the latest tailord version (0.2.4) from the AUR.

```
Operating System: Arch Linux 
KDE Plasma Version: 6.0.3
KDE Frameworks Version: 6.0.0
Qt Version: 6.6.3
Kernel Version: 6.8.2-zen2-1-zen (64-bit)
Graphics Platform: Wayland
Processors: 32 × 13th Gen Intel® Core™ i9-13900HX
Memory: 62.5 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: TUXEDO
Product Name: TUXEDO Stellaris Intel Gen5
```

AaronErhardt, 2024-04-01:
This is extremely likely a problem with the AUR package, over which I have no control. The system service and the GUI share a crate called "tailor_api" which defines the  JSON structure used for sending messages between those components. If you compile from git it is virtually impossible that you get a serialization error. The problem can only appear if you compile either GUI or the system service with an outdated version of that crate.

evertvorster, 2024-04-01:
Hmm, the sources checked out by tailord and tailor-gui are different. 
tailord checks out https://github.com/AaronErhardt/tuxedo-rs/archive/refs/tags/tailord-0.2.4.tar.gz

and tailor-gui checks out git+https://github.com/AaronErhardt/tuxedo-rs/#commit=fcca331

Are you suggesting that both packages should check out the same source?

AaronErhardt, 2024-04-01:
> Are you suggesting that both packages should check out the same source?

Yes, they always should. I've issued a new release for all packages with updated lockfiles, which should hopefully fix all the problems.

evertvorster, 2024-04-01:
OK, I'll notify the package maintainer.

evertvorster, 2024-04-03:
Just installed the newly updated Arch packages, and this issue has now been resolved.

closed 2024-04-03 by evertvorster