AaronErhardt/tuxedo-rs #70: Support for Stellaris 16 Gen5 (checklist?)

mutech, 2024-03-28:
I have one of these and trouble getting TCC to work under Arch, so I'm looking into getting tuxedo-rs up. I'll come back when I did.

My laptop is not on the list of supported devices. Ho do you test tuxedo-rs when you add it to the list? The reasons why I'm trying to install tuxedo-rs are:
- Get rid of the light show on the keyboard (it random decides to show off and rainbow me)
- Set charging to stationary mode to improve the battery lifetime
- Occassionally change a performance profile
- Occassionally change the GPU setting (intel/mixed/nvidia)

Do you have a check list that you're going through to see if everything is working as expected when you're testing?

AaronErhardt, 2024-03-28:
From the README:

> To find out more about the features supported by your device, you can install the `tailor_hwcaps` CLI-tool:
>
> ```sh
> cargo install tailor_hwcaps --git https://github.com/AaronErhardt/tuxedo-rs && tailor_hwcaps
> ```

If you can paste the output here, I can tell you what's going to work on your device if it's not already clear on its own.

Currently, setting the keyboard LEDs and the performance profile are supported. I don't know what we have to do to support "stationary mode" for charging. Regarding GPU, I'd recommend using something like CoreCtrl, it really doesn't make sense to reinvent the wheel for us.

mutech, 2024-04-06:
Here is the output of hwcaps:

[hwcaps.out.txt](https://github.com/AaronErhardt/tuxedo-rs/files/14892023/hwcaps.out.txt)

AaronErhardt, 2024-04-09:
It looks like the tuxedo-rs currently supports 2 fans and 127 LED devices. Tuxedo-rs is not quite optimized to handle that many LEDs in the UI, but in theory you can even set animations for each LED (which is likely a single key on your keyboard) individually.

Performance profiles are not supported on your device currently and would likely need a patch to work (but I don't have the hardware to work on this).

mutech, 2024-04-13:
You should contact Tuxedo, maybe they'll send you hardware? If so, it's probably a nice compensation for the development effort, these laptops are amazing.

I'm closing this issue. I would have rather offered a PR, but my rust skills are marginal and I don't have enough time to learn what's missing right now. Thanks for your work, Aaron, it looks great, even if it's not for me right now.

closed 2024-04-13 by mutech

mutech, 2024-04-13:
And yes, I can actually light up keys individually, how cool is that?!