AaronErhardt/tuxedo-rs #74: Minimum fan speeds a little too high at low temperatures.

evertvorster, 2024-04-09:
Hi there!

Using the fan control app here, and I notice a limitiation of the minimum fan speed that could use some refinement. 

Currently there is a slope built into the system which seems to start at 50C at 0 RPM, and then goes up to 100% at 100C.

While this does protect the CPU at high temperatures, There is absolutely no danger to the CPU below 80C, and enforcing any fan speed at all below 80C is artificially limiting the options to the user.

If you are running in powersave mode away on battery power, you don't want the fans to spin at all until about 70C. These laptop CPUs are quite happy at these temperatures, and it saves power to run fans slowly over a hot CPU.

What I would like to be able to do is to set fan speed to 0% up to 70C, then bump it up to 20%  at 70C and slope it to 40% at 80C.

From 80C to 95C the fans have to smoothly ramp from 40% to 100%

AaronErhardt, 2024-04-09:
Technically, there's no actual need for such limitations anyway, I just thought that might be a good limit to prevent thermal throttling and increase your hardware's lifetime slightly.

So it sounds reasonable to lift the limit to start with 0% from 75°C with 5% steps per degree, reaching 100% at 95°C. I want to keep it as a simple line without changes in slope, that's something you can do manually. What do you think?

evertvorster, 2024-04-10:
Your suggestion sounds just fine, and would allow me to get the fan profile I like plus protect hardware from throttling... 

On a side note, I think this hardware throttles in any case, because even at full fan speed the frequencies drop a little after a while....

PR #76 (2024-04-10): feat: Lift fan limit to start at 75°C
Fixes #74
closed 2024-04-10 by AaronErhardt