AaronErhardt/tuxedo-rs #82: Logic errors in fan control setup.

evertvorster, 2024-04-16:
Hi there!

When setting curves for fan control, there is a drop-down fan 1 / fan 2 selector that does not seem to do anything. 

At least, whatever is selected is not remembered, and the next time you open the fan control setting it defaults back to fan 1 if you had selected fan 2. 

Then, when setting up the profile, there is a setting for fan 1 and fan 2, and there you can select whatever profile you want for the fan to be controlled. 

It would be my suggestion to remove that drop-down selector, and save fan curves as fan curves only, and then apply that curve to whichever fan is selected in the profile setup. ;)

Also, which sensor activates/controls Fan 2? Are both fans controlled by CPU temperature? Fan 2 on my machine is the GPU fan, and should be controlled by the GPU temperature....

How are we going to resolve that?

AaronErhardt, 2024-04-16:
The drop down is only for a preview or rather "prenoise" and allows you to hear how load a certain fan spins up at a certain point in your profile (whenever you drag a data point, the fan will be set to the speed of that exact point for a brief moment). This is the same behavior as for the LED lights, which also have a short preview when you select a color.

> Also, which sensor activates/controls Fan 2? Are both fans controlled by CPU temperature? Fan 2 on my machine is the GPU fan, and should be controlled by the GPU temperature....

It should be the same device. The CPU temperature affects the CPU fan and the GPU temperature the GPU fan.

evertvorster, 2024-04-16:
Hi there!
Thank you for the explanation, it does make more sense now. 
In my case, after I had selected a point on the graph, the fan stays at the selected speed until I select another point.

It would be nice to not need a special description in order to understand this. 
Maybe move the drop-down to the right hand top, and put "Preview fan noise for " in front of it?

Unfortunately, I ran into a much more serious issue today, but I will make that a separate issue.