AaronErhardt/tuxedo-rs #83: Fan 2 not spinning up  automatically.

evertvorster, 2024-04-16:
I did a quick test with a graphic intensive application (DOOM on Steam) and not only did the fan not spin up, the GPU shut down due to temperature... which was very disconcerting to say the least. 
When I say the GPU fan did not spin up, I mean I could not hear it spinning, and could feel no air coming from that side of the laptop. The CPU fan was spinning fast, and I could feel hot air coming out of those vents. 

It would be really nice to have a dashboard to see the actual fan speeds and temperatures involved, but for now I can for certain say that my GPU fan does not spin up enough  automatically under load. 

With Tuxedo control center, I can monitor the actual CPU and GPU temperatures, and also how fast the fans are spinning. Unfortunately the control center does not work while tailord is active, so the monitoring of these critical data points are lost.

In any case, I can make the GPU fan (fan 2) spin up by selecting a point on the configuration graph while having fan 2 selected on the drop-down, so the fan control part is working, but it is not responding to changes in CPU/GPU temperature. 

As for which temperature sensor should drive the fan speed, there may not be a simple solution. It is possible to run either the CPU or the GPU hard without affecting the other much.  If both fans are controlled by the CPU temperature sensor, you could run into trouble if you were running an application that uses the GPU heavily, but not the CPU, and so the CPU would remain cool while the GPU temps would be running away. 

On my laptop, at least, the GPU and CPU are connected together, and each has has it's own fan. So, yes, they have an influence on each other, and it would be possible to use the GPU fan to help cool the CPU, and it is also possible to for the CPU fan to help cool the GPU. 

Maybe the most simple option would be to monitor both the GPU and CPU temperature, and use the highest of the two to drive the fans at the same speed? 

A more elegant option is to only spin up the GPU fan for the GPU up to a certain percentage of fan speed, and then when it becomes noisy anyways to start spinning up the CPU fan to help with the cooling. 
The same could be done for the CPU, and the GPU fan helping out when the CPU fan is becoming noisy anyways... 

The way the Tuxedo control center does it seems to have both fans spinning at the same speed all the time, but that seems to be a little wasteful on power, because if the CPU temps are lowish, the CPU fan should be able to keep it cool without the help of the GPU fan, and should be able to do so quietly.

AaronErhardt, 2024-04-21:
I think having a dashboard is a good idea for a new feature. For now you can help debugging this issue either by running `RUST_LOG=debug tailord` or using `tailor_hwcaps` repeatedly. I couldn't find any obvious bug in the code and logging the fan information should help us figuring out whether tailord actually sets the desired fan speed correctly.

evertvorster, 2024-04-21:
Hi there!

I made a little script to monitor my fan speeds:
```
#!/bin/bash
while [[ 0 -ne 1 ]]; do 
echo "Current fan speeds:"
tailor_hwcaps | grep -i fan
echo ""
echo "Press Ctrl+C to exit"
sleep 1;
clear
done
```

With this script, I could see that the two fans are working completely independently from each other. 
But, they were working, and after giving Fan 2 a profile where the fan would speed up faster with lower temperatures, I could see that it works normally. 

So, with a utility that hammers only the CPU, I can have the CPU fan going full speed, with the CPU sensor saying 99C, and the GPU having a temperature of 55C, and it's fan sitting at 20% because of that low temperature. 

Alternately, I could hammer the GPU on it's own, and have its fan sitting at 100% with a temperature of 85C, and the CPU fan idling at 30% with a temperature of 65C. 

We could close this ticket, but considering that on this laptop at least, the heat sinks for the GPU and CPU are tied together, it would make sense for there being some sort of co-operation between the two fans at high temps for either sensor.  

Which would be the best way to achieve this co-operation between the fans for either GPU or CPU sensor?

nagymathev, 2024-05-25:
It indeed would be really good to have a toggle where the fans would base their speed on the highest temperature unit. So if the GPU is really hot, it would make sense to have both fans spin up to try and cool it down since my machine also has an interconnected heat-sink.

I would love to give it a try, but I haven't fiddled with rust or hardware level programming much.

evertvorster, 2024-05-26:
I've let this percolate in the back of my brain for a while. 

There needs to be some kind of formula that ties the fan speeds together. 
As the demand for fan speed increases, it should have an increasing effect on the minimum fan speed of the other fan. 

I think the following formula would do the trick:
At 0%, One fan should have 0% influence on the other fan, at 100%, 100%. 

Let's make up a few scenarios, and see if this works. 
CPU 50% fan speed demanded. GPU at 0%.  
50% of 50 is 25. 
In this case, the GPU fan speed would be artificially raised to 25%, which seems fair. 
If it was spinning at 50% anyways, no change. 

Second Scenario:
Demanded speeds: CPU fan at 80%, GPU at 30%
80% of 80 is 64, so the minimum fan speed of the GPU would be raised to 64%

At 90% the minimum for the other fan should be 81%

So, that seems to work. 
Not having looked at the code, I don't know if this is possible without major changes. 
There should be a concept of demanded fan speed based on the user setting, and a modifier based on a minimum. There is already a minimum based solely on temperature, and that should stay in place. I think the minimum fan speed should be a "whichever is greater" variable. So, if the GPU is cool, but the CPU is baking, the GPU fan should be able to help out a little.