AaronErhardt/tuxedo-rs #84: Seemingly random kernel crashes, but only when running tailord...

evertvorster, 2024-04-26:
Hi there!

Please have a look at this kernel crash log. My laptop seems to crash quite a bit more often when running tailord. 
Does any of it look familiar?
```
Apr 26 17:40:41 Evert.Laptop kernel: BUG: kernel NULL pointer dereference, address: 00000000000006e9
Apr 26 17:40:41 Evert.Laptop kernel: #PF: supervisor read access in kernel mode
Apr 26 17:40:41 Evert.Laptop kernel: #PF: error_code(0x0000) - not-present page
Apr 26 17:40:41 Evert.Laptop kernel: PGD 0 P4D 0 
Apr 26 17:40:41 Evert.Laptop kernel: Oops: 0000 [#5] PREEMPT SMP NOPTI
Apr 26 17:40:41 Evert.Laptop kernel: CPU: 1 PID: 2754 Comm: systemd-journal Tainted: P      D    OE      6.8.7-arch1-1 #1 cb8440eaa48704794690ea311c777c18c4e95af9
Apr 26 17:40:41 Evert.Laptop kernel: Hardware name: TUXEDO PD5x_7xSNC_SND_SNE/PD5x_7xSNC_SND_SNE, BIOS 1.07.11SLS1 05/26/2023
Apr 26 17:40:41 Evert.Laptop kernel: RIP: 0010:simple_xattr_get+0x31/0xa0
Apr 26 17:40:41 Evert.Laptop kernel: Code: 00 00 41 56 49 89 ce 41 55 4c 8d 6f 08 41 54 49 89 d4 55 48 89 f5 53 48 89 fb 4c 89 ef e8 47 0f 97 00 48 8b 1b 48 85 db 74 1b <48> 8b 7b 18 48 89 ee e8 63 28 94 00 85 c0 78 27 74 2b 48 8b 5b 08
Apr 26 17:40:41 Evert.Laptop kernel: RSP: 0018:ffffba81c88c7ae0 EFLAGS: 00010206
Apr 26 17:40:41 Evert.Laptop kernel: RAX: 0000000000000800 RBX: 00000000000006d1 RCX: 0000000000000067
Apr 26 17:40:41 Evert.Laptop kernel: RDX: ffff9c596b954e80 RSI: ffffba81c88c7bb0 RDI: ffff9c5956541220
Apr 26 17:40:41 Evert.Laptop kernel: RBP: ffffba81c88c7bb0 R08: ffff9c596b954e80 R09: 0000000000000067
Apr 26 17:40:41 Evert.Laptop kernel: R10: ffffba81c88c7bb0 R11: 0000000000000002 R12: ffff9c596b954e80
Apr 26 17:40:41 Evert.Laptop kernel: R13: ffff9c5956541220 R14: 0000000000000067 R15: ffffba81c88c7bb0
Apr 26 17:40:41 Evert.Laptop kernel: FS:  0000741831151540(0000) GS:ffff9c68bf040000(0000) knlGS:0000000000000000
Apr 26 17:40:41 Evert.Laptop kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 26 17:40:41 Evert.Laptop kernel: CR2: 00000000000006e9 CR3: 000000021b794000 CR4: 0000000000f50ef0
Apr 26 17:40:41 Evert.Laptop kernel: PKRU: 55555554
Apr 26 17:40:41 Evert.Laptop kernel: Call Trace:
Apr 26 17:40:41 Evert.Laptop kernel:  <TASK>
Apr 26 17:40:41 Evert.Laptop kernel:  ? __die+0x23/0x70
Apr 26 17:40:41 Evert.Laptop kernel:  ? page_fault_oops+0x171/0x4e0
Apr 26 17:40:41 Evert.Laptop kernel:  ? exc_page_fault+0x7f/0x180
Apr 26 17:40:41 Evert.Laptop kernel:  ? asm_exc_page_fault+0x26/0x30
Apr 26 17:40:41 Evert.Laptop kernel:  ? simple_xattr_get+0x31/0xa0
Apr 26 17:40:41 Evert.Laptop kernel:  __vfs_getxattr+0x7f/0xb0
Apr 26 17:40:41 Evert.Laptop kernel:  do_getxattr+0xf0/0x180
Apr 26 17:40:41 Evert.Laptop systemd[1]: Started Journal Service.
Apr 26 17:40:41 Evert.Laptop kernel:  getxattr+0x91/0xd0
Apr 26 17:40:41 Evert.Laptop kernel:  path_getxattr+0x7b/0xd0
Apr 26 17:40:41 Evert.Laptop kernel:  do_syscall_64+0x83/0x170
Apr 26 17:40:41 Evert.Laptop kernel:  ? __do_sys_newfstat+0x6b/0x80
Apr 26 17:40:41 Evert.Laptop kernel:  ? __seccomp_filter+0x32c/0x4f0
Apr 26 17:40:41 Evert.Laptop kernel:  ? syscall_exit_to_user_mode+0x83/0x230
Apr 26 17:40:41 Evert.Laptop kernel:  ? do_syscall_64+0x90/0x170
Apr 26 17:40:41 Evert.Laptop kernel:  ? __memcg_slab_free_hook+0xef/0x140
Apr 26 17:40:41 Evert.Laptop kernel:  ? __x64_sys_close+0x3d/0x80
Apr 26 17:40:41 Evert.Laptop kernel:  ? kmem_cache_free+0x3b7/0x3e0
Apr 26 17:40:41 Evert.Laptop kernel:  ? syscall_exit_to_user_mode+0x83/0x230
Apr 26 17:40:41 Evert.Laptop kernel:  ? do_syscall_64+0x90/0x170
Apr 26 17:40:41 Evert.Laptop kernel:  ? exc_page_fault+0x7f/0x180
Apr 26 17:40:41 Evert.Laptop kernel:  entry_SYSCALL_64_after_hwframe+0x78/0x80
Apr 26 17:40:41 Evert.Laptop kernel: RIP: 0033:0x74183152256e
Apr 26 17:40:41 Evert.Laptop kernel: Code: 37 0d 00 f7 d8 64 89 02 ba ff ff ff ff eb d5 67 e8 07 15 01 00 0f 1f 80 00 00 00 00 f3 0f 1e fa 49 89 ca b8 c0 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 92 37 0d 00 f7 d8 64 89 01 48
Apr 26 17:40:41 Evert.Laptop kernel: RSP: 002b:00007ffcdbefa6e8 EFLAGS: 00000246 ORIG_RAX: 00000000000000c0
Apr 26 17:40:41 Evert.Laptop kernel: RAX: ffffffffffffffda RBX: 0000000000000067 RCX: 000074183152256e
Apr 26 17:40:41 Evert.Laptop kernel: RDX: 00005ad44ae5f430 RSI: 00005ad44a4013f8 RDI: 00005ad44ae5d4c0
Apr 26 17:40:41 Evert.Laptop kernel: RBP: 00005ad44ae5f430 R08: 0000000000000070 R09: 0000000000000001
Apr 26 17:40:41 Evert.Laptop kernel: R10: 0000000000000067 R11: 0000000000000246 R12: 00005ad44ae5d4c0
Apr 26 17:40:41 Evert.Laptop kernel: R13: 00000000ffffff9c R14: 0000000000000000 R15: 00005ad44a4013f8
Apr 26 17:40:41 Evert.Laptop kernel:  </TASK>
Apr 26 17:40:41 Evert.Laptop kernel: Modules linked in: tun dummy ccm rfcomm snd_hrtimer xe drm_gpuvm drm_exec gpu_sched drm_suballoc_helper drm_ttm_helper cmac algif_hash algif_skcipher af_alg xt_pkttype snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel snd_sof_intel_hda_mlink soundwire_cadence snd_sof_intel_hda snd_sof_>
Apr 26 17:40:41 Evert.Laptop kernel:  videobuf2_memops ghash_clmulni_intel btusb sha512_ssse3 videobuf2_v4l2 snd_hda_intel btrtl sha1_ssse3 btintel snd_intel_dspcfg snd_intel_sdw_acpi aesni_intel libarc4 hid_multitouch videodev btbcm snd_hda_codec ptp hid_generic nvidia_drm(POE) btmtk crypto_simd nvidia_modeset(POE) nvidia(POE) i915 p>
Apr 26 17:40:41 Evert.Laptop kernel:  processor_thermal_mbox idma64 intel_pmc_core roles video clevo_acpi(OE) i2c_hid_acpi tuxedo_io(OE) int3400_thermal int3403_thermal tuxedo_keyboard(OE) intel_vsec i2c_hid tuxedo_compatibility_check(OE) intel_hid led_class_multicolor pmt_telemetry pinctrl_alderlake acpi_thermal_rel int340x_thermal_z>
Apr 26 17:40:41 Evert.Laptop kernel: Unloaded tainted modules: tuxedo_nb02_nvidia_power_ctrl(OE):1
Apr 26 17:40:41 Evert.Laptop kernel: CR2: 00000000000006e9
Apr 26 17:40:41 Evert.Laptop kernel: ---[ end trace 0000000000000000 ]---
Apr 26 17:40:41 Evert.Laptop kernel: RIP: 0010:simple_xattr_get+0x31/0xa0
Apr 26 17:40:41 Evert.Laptop kernel: Code: 00 00 41 56 49 89 ce 41 55 4c 8d 6f 08 41 54 49 89 d4 55 48 89 f5 53 48 89 fb 4c 89 ef e8 47 0f 97 00 48 8b 1b 48 85 db 74 1b <48> 8b 7b 18 48 89 ee e8 63 28 94 00 85 c0 78 27 74 2b 48 8b 5b 08
Apr 26 17:40:41 Evert.Laptop kernel: RSP: 0018:ffffba81c4f639c8 EFLAGS: 00010206
Apr 26 17:40:41 Evert.Laptop kernel: RAX: 0000000000000000 RBX: 00000000000006d1 RCX: 0000000000000067
Apr 26 17:40:41 Evert.Laptop kernel: RDX: ffff9c59f6bb2c80 RSI: ffffba81c4f63a98 RDI: ffff9c5956541220
Apr 26 17:40:41 Evert.Laptop kernel: RBP: ffffba81c4f63a98 R08: ffff9c59f6bb2c80 R09: 0000000000000067
Apr 26 17:40:41 Evert.Laptop kernel: R10: ffffba81c4f63a98 R11: 0000000000000002 R12: ffff9c59f6bb2c80
Apr 26 17:40:41 Evert.Laptop kernel: R13: ffff9c5956541220 R14: 0000000000000067 R15: ffffba81c4f63a98
Apr 26 17:40:41 Evert.Laptop kernel: FS:  0000741831151540(0000) GS:ffff9c68bf040000(0000) knlGS:0000000000000000
Apr 26 17:40:41 Evert.Laptop kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 26 17:40:41 Evert.Laptop kernel: CR2: 00000000000006e9 CR3: 000000021b794000 CR4: 0000000000f50ef0
Apr 26 17:40:41 Evert.Laptop kernel: PKRU: 55555554
Apr 26 17:40:41 Evert.Laptop kernel: note: systemd-journal[2754] exited with irqs disabled
Apr 26 17:40:41 Evert.Laptop kernel: note: systemd-journal[2754] exited with preempt_count 1
Apr 26 17:40:41 Evert.Laptop systemd[1]: systemd-journald.service: Scheduled restart job, restart counter is at 6.
Apr 26 17:40:41 Evert.Laptop systemd[1]: Starting Journal Service...
Apr 26 17:40:41 Evert.Laptop systemd-journald[2760]: Collecting audit messages is disabled.
Apr 26 17:40:41 Evert.Laptop systemd-journald[2760]: File /var/log/journal/a9d98e830d8d44b3904796e9a59ede5c/system.journal corrupted or uncleanly shut down, renaming and replacing.
Apr 26 17:40:41 Evert.Laptop systemd-journald[2760]: Journal started
```

AaronErhardt, 2024-04-27:
Tailord is just in user space and despite interacting with the kernel, it should now be able to cause a panic. Also, I don't even have the module in question (tuxedo_nb02_nvidia_power_ctl) on my system, so I couldn't even debug this if I wanted.
I suggest to report this to tuxedo-drivers instead: https://github.com/tuxedocomputers/tuxedo-drivers?tab=readme-ov-file

evertvorster, 2024-04-27:
Hi there, Aaron. 
Ah, here's the rub.. this is not Tuxedo hardware, and they are supremely non-interested in troubleshooting any issues with it.

I guess I will just lurk in the shadows and see if any changes happen to that module and try again. 
This would all have been so much simpler if I could get my hands on a BIOS for a Tux Gemini Gen 2, because then it would be indistinguishable from a machine that they sell. Again, they would not even sell me a BIOS...

evertvorster, 2024-04-27:
Hi there again!

I managed to run into this issue again without running tailord, which confirms your position and vindicates tailord. 
There may be an issue with nVidia's 550 driver, and I am in the process of downgrading now. 
My apologies for cluttering your project's issues with my stuff, I'll close this issue now. 

Thanks again just for your comment, though, it pointed me in a direction to troubleshoot this issue forther.

closed 2024-04-27 by evertvorster