AaronErhardt/tuxedo-rs #86: Poor dGPU performance when enabling tuxedo-rs option in NixOS

nagymathev, 2024-05-24:
Hey, I've noticed that when I enable `tuxedo-rs` in my configuration the performance of my dGPU takes a huge hit, apps get insane screen-tear, way slower and laggier than the iGPU, almost as if it would stop working or just barely functioning. I have no idea what could cause such things to occur other than enabling the option in my NixOS configuration. I have the `tuxedo_keyboard` enabled as well. I have the TUXEDO Stellaris Gen 5 AMD.

Any help debugging or resolving this issue would be much appreciated!

AaronErhardt, 2024-05-24:
You might check the fan curve of your GPU. Also the hwcals utility as described on the README can help you monitoring the temperature. I assume it's a thermal issue and the default fan curve is not fast enough for you GPU.

nagymathev, 2024-05-24:
It's actually not a thermal issue, after seeing the thermal results and adjusting fan curves accordingly, I have come across a weird *bug*? It might just be my system, but when I set the profile anything other than the default even if the backlight and the fan controls are default, and restart, so that the system starts with the custom profile, that's when the performance hit occurs. If I then switch the profile back to default everything seems to work as intended, *even after switching back the profile to the non-default one*. But if I have the system boot with a non default profile I have to set it to default and back every time. So I am really confused now. I'm not sure if something is not being loaded correctly or else.

Any ideas, or suggestions on that? My mind is just blown away.

AaronErhardt, 2024-05-25:
That's odd because the control loop doesn't even know when it run's a default profile. It just loads the default values in this case and you can even overwrite them. So if you create a new profile based on the default one, it should not change anything...