AaronErhardt/tuxedo-rs #87: How to set/use fan profile

shot-codes, 2024-06-04:
```
OS: NixOS 24.11.20240529.ad57eef (Vicuña) x86_64
Host: TUXEDO InfinityBook Pro Gen8 (MK2) (Standard)
Kernel: Linux 6.6.32
````
I enable tuxedo-rs, with tailor_gui, through my NixOS config as per the documentation. I am able to launch tailor_gui and create new profiles and fan profiles. However, I am unsure of how to either assign my created fan profile, or "link" it to a profile. Furthermore, my created Profiles are not persistent, the fan profiles are.

Logs; Launching tailor_gui -> creating a Profile -> opening my custom fan profile -> saving custom fan profile.
```
2024-06-04T08:38:39.491063Z  INFO perform: zbus::handshake: new
2024-06-04T08:38:39.491088Z  INFO perform: zbus::handshake: enter
2024-06-04T08:38:39.491112Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: new
2024-06-04T08:38:39.491117Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2024-06-04T08:38:39.491125Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2024-06-04T08:38:39.491129Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: enter
2024-06-04T08:38:39.491132Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: exit
2024-06-04T08:38:39.491135Z  INFO perform:write_command{command=Auth(Some(External), Some([49, 48, 48, 48]))}: zbus::handshake: close time.busy=10.4µs time.idle=12.8µs
2024-06-04T08:38:39.491143Z  INFO perform:read_command: zbus::handshake: new
2024-06-04T08:38:39.491146Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491149Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491152Z  INFO perform: zbus::handshake: exit
2024-06-04T08:38:39.491275Z  INFO perform: zbus::handshake: enter
2024-06-04T08:38:39.491282Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491293Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491296Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491298Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491301Z  INFO perform:read_command: zbus::handshake: close time.busy=16.7µs time.idle=141µs
2024-06-04T08:38:39.491307Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: new
2024-06-04T08:38:39.491310Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2024-06-04T08:38:39.491322Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2024-06-04T08:38:39.491326Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: enter
2024-06-04T08:38:39.491328Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: exit
2024-06-04T08:38:39.491331Z  INFO perform:write_command{command=NegotiateUnixFD}: zbus::handshake: close time.busy=14.5µs time.idle=9.39µs
2024-06-04T08:38:39.491336Z  INFO perform:read_command: zbus::handshake: new
2024-06-04T08:38:39.491340Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491344Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491347Z  INFO perform: zbus::handshake: exit
2024-06-04T08:38:39.491433Z  INFO perform: zbus::handshake: enter
2024-06-04T08:38:39.491442Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491449Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491452Z  INFO perform:read_command: zbus::handshake: enter
2024-06-04T08:38:39.491455Z  INFO perform:read_command: zbus::handshake: exit
2024-06-04T08:38:39.491457Z  INFO perform:read_command: zbus::handshake: close time.busy=13.2µs time.idle=107µs
2024-06-04T08:38:39.491462Z  INFO perform:write_command{command=Begin}: zbus::handshake: new
2024-06-04T08:38:39.491466Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2024-06-04T08:38:39.491475Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2024-06-04T08:38:39.491479Z  INFO perform:write_command{command=Begin}: zbus::handshake: enter
2024-06-04T08:38:39.491482Z  INFO perform:write_command{command=Begin}: zbus::handshake: exit
2024-06-04T08:38:39.491485Z  INFO perform:write_command{command=Begin}: zbus::handshake: close time.busy=12.7µs time.idle=10.2µs
2024-06-04T08:38:39.491490Z  INFO perform: zbus::handshake: exit
2024-06-04T08:38:39.491493Z  INFO perform: zbus::handshake: enter
2024-06-04T08:38:39.491495Z  INFO perform: zbus::handshake: exit
2024-06-04T08:38:39.491497Z  INFO perform: zbus::handshake: close time.busy=196µs time.idle=243µs
2024-06-04T08:38:39.491567Z  INFO socket reader: zbus::socket_reader: new
2024-06-04T08:38:39.491573Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.491580Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.491837Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.491919Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.492556Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.492609Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.493103Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.493137Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.493604Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.493639Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.493652Z  INFO tailor_gui::state: No performance handler available: Bus response error: `org.freedesktop.DBus.Error.Failed: No performance profile handler available`
2024-06-04T08:38:39.493975Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.494025Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.494575Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:38:39.494618Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:38:39.494891Z  INFO socket reader: zbus::socket_reader: enter
[... 151 lines cut ...]
2024-06-04T08:39:02.759151Z  INFO update_with_view{input=Enabled(DynamicIndex { inner: 1 }) component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: new
2024-06-04T08:39:02.759185Z  INFO update_with_view{input=Enabled(DynamicIndex { inner: 1 }) component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: enter
2024-06-04T08:39:02.759207Z  INFO update_with_view{input=Enabled(DynamicIndex { inner: 1 }) component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: exit
2024-06-04T08:39:02.759224Z  INFO update_with_view{input=Enabled(DynamicIndex { inner: 1 }) component="tailor_gui::components::profiles::Profiles" id="Profiles"}: relm4::component::sync::builder: close time.busy=22.0µs time.idle=50.3µs
2024-06-04T08:39:02.760127Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:39:02.760310Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:39:02.761556Z  INFO socket reader: zbus::socket_reader: enter
2024-06-04T08:39:02.761674Z  INFO socket reader: zbus::socket_reader: exit
2024-06-04T08:39:02.765147Z  INFO update_with_view{input=UpdateProfiles(["default"]) component="tailor_gui::components::led_list::LedList" id="0x7ffe44cea700"}: relm4::component::sync::builder: new
2024-06-04T08:39:02.765182Z  INFO update_with_view{input=UpdateProfiles(["default"]) component="tailor_gui::components::led_list::LedList" id="0x7ffe44cea700"}: relm4::component::sync::builder: enter
2024-06-04T08:39:02.765966Z  INFO update_with_view{input=UpdateProfiles(["default"]) component="tailor_gui::components::led_list::LedList" id="0x7ffe44cea700"}: relm4::component::sync::builder: exit
2024-06-04T08:39:02.765987Z  INFO update_with_view{input=UpdateProfiles(["default"]) component="tailor_gui::components::led_list::LedList" id="0x7ffe44cea700"}: relm4::component::sync::builder: close time.busy=783µs time.idle=57.7µs
2024-06-04T08:39:02.766026Z  INFO update_with_view{input=UpdateProfiles(["default", "speed"]) component="tailor_gui::components::fan_list::FanList" id="0x7ffe44cea648"}: relm4::component::sync::builder: new
2024-06-04T08:39:02.766035Z  INFO update_with_view{input=UpdateProfiles(["default", "speed"]) component="tailor_gui::components::fan_list::FanList" id="0x7ffe44cea648"}: relm4::component::sync::builder: enter
[... cut ...]
2024-06-04T08:39:02.772285Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.FileNotFound: Couldn't find profile `speed``") component="tailor_gui::app::App" id="0x7ffe44cea678"}: relm4::component::sync::builder: exit
2024-06-04T08:39:02.772289Z  INFO update_with_view{input=AddError("Bus response error: `org.freedesktop.DBus.Error.FileNotFound: Couldn't find profile `speed``") component="tailor_gui::app::App" id="0x7ffe44cea678"}: relm4::component::sync::builder: close time.busy=15.3µs time.idle=9.51µs
```

shot-codes, 2024-06-04:
https://github.com/AaronErhardt/tuxedo-rs/assets/43202406/18b26237-177b-4c53-97b0-6c094a84197e

AaronErhardt, 2024-06-04:
The log messages were already reported in https://github.com/AaronErhardt/tuxedo-rs/issues/71, which was fixed on git, but not released yet. You should also be able to fix this by removing the existing config in `/etc/tailord` and restarting tailord.

shot-codes, 2024-06-04:
Thanks for the response! And thanks for maintaining to the benefit of all tuxedo users <3. Will give this a shot later and close if I am successful. Feel free to close anyways if you think it's appropriate.

shot-codes, 2024-06-04:
Removing /etc/tailord did the trick, thanks!

closed 2024-06-04 by shot-codes