AaronErhardt/tuxedo-rs #103: Bug with Tuxedo Pulse 14 Gen 3

gabyx, 2024-06-19:
When I use `tailor_hdwcap` I get on Tuxedo Pulse 14 Gen 3:

```shell
RUST_BACKTRACE=1 tailor_hwcaps
thread 'main' panicked at /build/tuxedo-rs-0.3.1-vendor.tar.gz/sudo/src/lib.rs:141:48:
failed to execute child: Os { code: 2, kind: NotFound, message: "No such file or directory" }
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: core::result::unwrap_failed
   3: core::result::Result<T,E>::expect
   4: sudo::with_env
   5: tailor_hwcaps::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```
Could we have some better error message to see whats going wrong. 
I am using the Nix install with version: 0.3.1

AaronErhardt, 2024-06-19:
This panic does not originate in tuxedo-rs but rather in the `sudo` crate ([precisely here](https://docs.rs/sudo/0.6.0/src/sudo/lib.rs.html#141)). Because `hwcaps` needs to read data from sysfs and ioctl, it needs elevated privileges which is done with the `sudo` crate.

Since the sudo binary is not the only commonly used solution for gaining elevated privileges anymore, we should migrate to another crate, but AFAIK there is no alternative at the moment.

gabyx, 2024-06-21:
That means I basically can run it with sudo and it should work, I should try again. :)