AaronErhardt/tuxedo-rs #106: Boosting not working on Aura 15 Gen1

yukijoou, 2024-07-19:
Hi, when switching from TCC to tailord, my CPU doesn't boost anymore. It oscillates from 1397MHz to 1400MHz according to htop. Any way to fix this?

Running on archlinux, latest everything.

AaronErhardt, 2024-07-19:
Have you checked your power profile? Also, have you tried `tailor_hwcaps` (see README)?

yukijoou, 2024-07-19:
Yes, I'm using the same power profile both on tcc and tailord.

Here's the output from hwcaps:

```
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "clevo_acpi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[ERR]   Model ID: NotAvailable
[OK]    Available ODM performance profiles: ["quiet", "power_saving", "entertainment", "performance"]
[OK]    Default ODM performance profile: "performance"
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [54]
[OK]    Fan speeds [%]: [31]
[OK]    Fan min speed [%]: 20
[OK]    Webcam enabled: true
[INFO]  TDP control is not available
[OK]    Number of LED devices: 1
[OK]    LED device number: 0
[OK]    LED device name: "platform:tuxedo_keyboard"
[OK]    LED device function: "kbd_backlight"
[OK]    LED mode: Rgb
[OK]    LED device color: Ok(Color { r: 146, g: 110, b: 228 })
```
I tried making the fan run at full blast to make sure it wasn't thermal throttling, and that didn't help. I did see it reach 2GHz after restarting tailord though, but still nowhere near the 4GHz I can get with tcc.

AaronErhardt, 2024-07-19:
Mhh, maybe this is related to this: https://github.com/AaronErhardt/tuxedo-rs/issues/36#issuecomment-1817541701

yukijoou, 2024-07-19:
I did start tailord after stopping tccd, without restarting though, so I don't know if it's really the same thing.

yukijoou, 2024-07-19:
update: I did try restarting my PC and not touching tccd at all - seems like it's back to being stuck between 1397MHz and 1400MHz, so that bug is definitely *part* of the issue. The boosting thing seems to be another issue though.

yukijoou, 2024-07-23:
okay, I may have figured out some things :)

it seems like I can get boosting to work by setting `/sys/devices/system/cpu/cpu*/cpufreq/boost` to `1`[^1], and selecting any governor but `powersave` (I found `ondemand` worked best for me)  for `/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`. Then, setting the "performance profile" to "entertainment" or "performance" in Tailor makes the CPU boost when active.

I don't recommend doing this, because on power it seems to be boosting very aggressively, meaning the fan can ramp up if you open htop or look at the laptop the wrong way :/ This also seems to make the CPU run hotter even when it's not boosting. I guess it's sending more power to it? I really don't know, I feel like something isn't adding up in the stats I'm seeing, and I'm siding with faulty hardware more and more.

My results with `/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor` set to `ondemand` and using the "entertainment" power profile:

* On battery:
  * CPU idles at \~45C, 1400MHz, fan runs at \~10%
  * Under a 100% load on all cores, the CPU runs at ~60C, 2000MHz to 2500MHz (so boosting, but not to "full capacity"), fan runs at ~35%
* Plugged in:
  * CPU idles at 55\~60C (constantly fluctuating), 1400MHz, fan runs at ~30%
  * Under a near 100% load on all cores, the CPU runs at \~90C, 3500mHz to 42000Mhz (seems like not all cores can hit 4.2Ghz at once), fan runs at 65\~70%.

It seems like it reaches 80~85C as soon as a single core gets to 4.2GHz. The main issue is that it seems to boost as soon as it gets a somewhat high cpu load, even if it's just for a few seconds, making the fans ramp up and down constantly due to background activity.

I'm not sure tailord can do much about this, it seems to be more down to the laptop not being great at handling CPU load. I do wonder though if there would be a way to automatically change the governor for *some* of the cores. I tried setting half of my cores to `powersave` and that seems to improve thermals somewhat.

[^1]: After rebooting and more testing, it seems like that's not necessary.