AaronErhardt/tuxedo-rs #116: Documentation on strict CLI usage

anthonymesa, 2024-10-05:
I have installed this on Pop!_OS 22.04 LTS on an Eluktronics Max 17 without issue, and it seems to control the keyboard and fans appropriately. My primary interest in using tuxedo-rs is to be able to manage and control my fan/keyboard via the CLI. I am not interested in going through the hoops to get gtk4 installed on 22, as I didn't plan to use GUI functionality anyways. 

That said, I do realize that creation of profiles seems to be something that happens in GUI only. Is there a guide or at least a robust example of a profile config available so that I may work only with the CLI side of things?

It seems that with documentation, users could write their own profiles and use the CLI more robustly as is. I would be interested in helping create this documentation and possibly trying to help expand the functionality of the CLI program, but I say that as a very new rustacean.

AaronErhardt, 2024-10-05:
> Is there a guide or at least a robust example of a profile config available so that I may work only with the CLI side of things?

There are the configuration files in `/etc/tailord` as an example and also the code is pretty self documenting: [profile](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailor_api/src/profile.rs), [keyboard colors](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailor_api/src/color.rs#L20) and [fan](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailor_api/src/fan.rs).

> I would be interested in helping create this documentation and possibly trying to help expand the functionality of the CLI program, but I say that as a very new rustacean.

I'd be happy to help you with that. I'm quite busy currently and don't use the CLI myself, but I see that this would be useful for others.

anthonymesa, 2024-10-05:
Thank you for your swift response. I did find the profile files you mentioned in /etc/tailord, and I was able to figure out how to create and apply fan profiles but the led profiles and performance profiles are not covered fully:

profiles/default.json
```
{
  "fans": [
    "default"
  ],
  "leds": [],
  "performance_profile": null
}
```

I had figured that leds would be handled similarly to Fans, but of course I have learned that there is more data expected in the led config, and the performance_profile. Given I don't have access to the gui version of this application and can not generate a profile that shows each of these settings fleshed out, would you be able to generate and upload a profile/config that was created with the gui that shows all parts customized?

I will continue to peruse the code and see if I cant put a merge request together for some documentation as I figure things out, but an already generated configuration file would speed up this process.

Also, if you are a user reading this, and you have a config file generated, please feel free to upload your example.

AaronErhardt, 2024-10-05:
Here's my config folder; I hope this helps you: [tailord.zip](https://github.com/user-attachments/files/17267299/tailord.zip)