AaronErhardt/tuxedo-rs #117: TUXEDO InfinityBook Pro AMD Gen9 support

SaltBasket-3702, 2024-10-12:
Just wanted to add feedback.  I can tell that fan control works indeed.  If there's anythink else I can do, tell me.

[tailord 0.2.5-1 from AUR](https://aur.archlinux.org/packages/tailord)
```
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "uniwill_wmi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Model ID: "26"
[OK]    Available ODM performance profiles: []
[ERR]   Default ODM performance profile: DevNotAvailable
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [41]
[OK]    Fan speeds [%]: [0]
[OK]    Fan min speed [%]: 20
[INFO]  Webcam control is not available
[ERR]   number_of_tdp_devices: DevNotAvailable
[ERR]   tdp_descriptors: DevNotAvailable
[OK]    tdps: []
[OK]    max_tdps: []
[OK]    min_tdps: []
[OK]    Number of LED devices: 0
```

ftelnov, 2024-12-09:
@SaltBasket-3702 Hello! You should have been using your TUXEDO laptop for a while now, could you please elaborate a bit on your experience? Is "TUXEDO InfinityBook Pro AMD Gen9" support decent on nixos, do you have any problems? Does tuxedo-rs solve your problems? Maybe you use something else these days? What `tuxedo-rs` version do you use? Thanks a lot in advance for any information!!

tnibler, 2025-01-02:
Do charging profiles work for you? I've tried writing to `/sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profile` but it always stays at "stationary".

XBagon, 2025-01-12:
I have the same except the ODM error:
```
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "uniwill_wmi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Model ID: "26"
[OK]    Available ODM performance profiles: ["power_save", "enthusiast", "overboost"]
[OK]    Default ODM performance profile: "enthusiast"
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [35]
[OK]    Fan speeds [%]: [9]
[OK]    Fan min speed [%]: 20
[INFO]  Webcam control is not available
[ERR]   number_of_tdp_devices: DevNotAvailable
[ERR]   tdp_descriptors: DevNotAvailable
[OK]    tdps: []
[OK]    max_tdps: []
[OK]    min_tdps: []
[OK]    Number of LED devices: 0
```

So the tdp stuff is supposed to be not available?

Also on random restarts (also this time) I have the issue of the fans running quiet loud/fast, without any load or temperature. Should 9% be so very noticeable?

AaronErhardt, 2025-01-13:
> Should 9% be so very noticeable?

I don't think so. Actually, at 9% the fans shouldn't run at all because the minimum fan speed is 20%. So something is wrong with the reporting of the driver, assuming that value was shown while the fans were loud (because the fan speed can change quickly)

XBagon, 2025-01-13:
Ah I forgot to mention the detail, that the fans won't ever stop until I restart my system. So yes the values are from when the fan was running. Actually I'm not quite sure, but I'm imagining that only the left fan is running fully.
When I adjust the fan curve in Tailor, the preview works, and runs only on the right fan.
Is that supposed to be that way and why is there only 1 fan detected?

AaronErhardt, 2025-01-14:
> Is that supposed to be that way and why is there only 1 fan detected?

No, certainly not. The logic from userspace (tuxedo-rs) is quite straight-forward as well [(here)](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tuxedo_ioctl/src/hal/uniwill.rs#L33C24-L33C43). The only reason for this behavior can only be a bug in the kernel driver or new kernel interfaces not used by tuxedo-rs yet. Since there is no documentation for the kernel driver the only option might be to reverse engineer the code from TUXEDO's control center.

mindrunner, 2025-02-01:
Can confirm the left-fan issue on AMD Gen9

pstark-code, 2025-03-27:
I get a slightly different output:

```
user@host$ cargo install tailor_hwcaps --git https://github.com/AaronErhardt/tuxedo-rs && sudo "$HOME/.cargo/bin/tailor_hwcaps"
    Updating git repository `https://github.com/AaronErhardt/tuxedo-rs`
     Ignored package `tailor_hwcaps v0.2.6 (https://github.com/AaronErhardt/tuxedo-rs#e80bdc72)` is already installed, use --force to override
warning: be sure to add `/home/pstark/.cargo/bin` to your PATH to be able to run the installed binaries
[OK]    Module version: "0.3.9\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Device interface ID: "uniwill_wmi\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"
[OK]    Model ID: "26"
[OK]    Available ODM performance profiles: ["power_save", "enthusiast", "overboost"]
[OK]    Default ODM performance profile: "enthusiast"
[OK]    Number of fans: 1
[OK]    Fan temperatures [°C]: [56]
[OK]    Fan speeds [%]: [13]
[OK]    Fan min speed [%]: 20
[INFO]  Webcam control is not available
[ERR]   number_of_tdp_devices: DevNotAvailable
[ERR]   tdp_descriptors: DevNotAvailable
[OK]    tdps: []
[OK]    max_tdps: []
[OK]    min_tdps: []
[OK]    Number of LED devices: 0
[OK]    Available charging profiles: ["high_capacity", "balanced", "stationary"]
[OK]    Current charging profile: "high_capacity"
[INFO]  Charging priority control is not available
[INFO]  Charge control for start/end thresholds is not available
```

I run NixOS, but I used a freshly compiled tailor_hwcaps in the output above. Let me know what info I can provide to help. 

I also have a fan issue, but a different one. I will create a separate issue, as soon as I make sure that tailord is actually causing the problem.

bdudelsack, 2025-05-25:
See https://github.com/AaronErhardt/tuxedo-rs/issues/151

bdudelsack, 2025-05-26:
> Ah I forgot to mention the detail, that the fans won't ever stop until I restart my system. So yes the values are from when the fan was running. Actually I'm not quite sure, but I'm imagining that [...]

Reloading the kernel driver seems to help too. Maybe there is some race condition? Should `clevo_wmi` be actually loaded?

```
tuxedo_io              24576  1
tuxedo_keyboard       118784  3 tuxedo_io,uniwill_wmi,clevo_wmi
tuxedo_compatibility_check    12288  1 tuxedo_keyboard
```

XBagon, 2025-05-29:
> Reloading the kernel driver seems to help too. Maybe there is some race condition? Should `clevo_wmi` be actually loaded?
>
> ```
> tuxedo_io              24576  1
> tuxedo_keyboard       118784  3 [...]

I learned that just putting my system to sleep and waking it up again, works as well. Probably that does reload the kernel driver?

mindrunner, 2025-06-03:
As far as I remember, I had the same issue on `Arch`. I only depend on my Laptop when traveling. Happy to help debugging this issue if needed.

bdudelsack, 2025-06-03:
My last test showed that you need only to reload the `uniwill_wmi` driver to stop the FAN from 100% spinning. So i assume something goes wrong in the initialization of the driver. I will try to go deeper.

XBagon, 2025-06-21:
> My last test showed that you need only to reload the `uniwill_wmi` driver to stop the FAN from 100% spinning. So i assume something goes wrong in the initialization of the driver. I will try to go deeper.

I tried this with
```
sudo rmmod uniwill_wmi
sudo modprobe uniwill_wmi
```
which is hopefully how you do this correctly :D?

and it does indeed stop the fan, but then it starts going faster again to some point, which feels less than the 100%, but it still stays there for seemingly no reason.. putting the system to sleep, and waking it up again doesn't bring the fan back.
But that could also be a problem, under heavy load my laptop just got very hot and I think the left fan didn't work at all.