AaronErhardt/tuxedo-rs #126: How to set the thermal source of a fan?

TheRedstoneDEV-DE, 2024-11-10:
Hello, I have a TUXEDO Gemini 15 Gen 2 and with my current setup on NixOS only one fan ramps up if the CPU gets hot and the second one only runs when the GPU gets hot. The problem being that the i9-13900HX gets really hot and would benefit from the second fan also running, which was the default configuration under TuxedoOS with TCC (basically the fans speeds were mirrored). As I didn't find a way in tuxedo-rs to set the thermal source of the second fan also to the CPU and not the GPU, I just wanted to ask if there is a way to set it and if it would be an option to implement it.
I hope this isn't a duplicate of another issue or a topic, that was already solved.
Thanks in advance.

AaronErhardt, 2024-11-10:
Currently, fans are completely on their own. Their read their only their own temperatures. It would be possible to implement some kind of dependency regarding the fan temperatures as you have suggested, but I would not be able to test this on my laptop, which makes it more difficult for me to work on this.

TheRedstoneDEV-DE, 2024-11-12:
You could give me instructions, and I can test it on my laptop (I just need it occasionally as a mobile workstation, so it isn't that bad if something breaks). I also have some basic Rust knowledge, so if there are any problems I might also be able to fix them or assist with something.

AaronErhardt, 2024-11-13:
Essentially, all the logic is abstracted into runtimes. Each fan or LED has its own runtime that runs asynchronously. Basically, the current implementation of a fan runtime [gets a fan device handle](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailord/src/fancontrol/mod.rs#L29C1-L29C33) and then updates the temperature periodically. When there are multiple fans, each fan has its own, separated runtime. You don't need to change that concept necessarily, but you'd have to give every fan runtime all device handles, so that each runtime can read the temperature of other fans as well. Then, you need to adjust the fan profile to be more than just [`Vec<FanProfilePoint>`](https://github.com/AaronErhardt/tuxedo-rs/blob/main/tailor_api/src/fan.rs), but introduce a struct that also stores, which other fan temperature should be considered (if any at all, so that should optional). That is a breaking change (because it changes the on-disk format), but we can keep backwards comparability quite easily, I think.

Let me know if you need more pointers, I know it can be somewhat confusing when looking at code other people wrote.